introlab/odas #121: Respeaker core v2.0

dinergumeno, 2018-10-09:
I would know if will you launch some configuration compatible with the Respeaker core v2.0?

FrancoisGrondin, 2018-10-10:
I think the best thing to do is that I'll contact ReSpeaker to get a board

doumdi, 2018-10-10:
Can I have one to for IntRoLab? ;)

dinergumeno, 2018-10-10:
People, I found a config file here in your repository that works well [https://github.com/introlab/odas/pull/72/commits/0cfed0a6f230ddb9ff7be61698ab4b169a27f775]

xiongyihui, 2018-10-18:
@doumdi Can you share your email? I will send you one.

doumdi, 2018-10-19:
@xiongyihui Please contact me at [dominic.letourneau@usherbrooke.ca](mailto:dominic.letourneau@usherbrooke.ca). Thanks!

janzuiderveld, 2019-01-25:
@xiongyihui @SethWelday Could you please send the config file for the respeaker core v2.0 to me too? Or if you have a specific config availaible for the 6-mic array for rpi, that would be amazing.

My email is: 
janzuiderveld@gmail.com

Thanks (:

xiongyihui, 2019-01-28:
@janzuiderveld You can find these configurations on my repository https://github.com/xiongyihui/odas/tree/master/config/odaslive

zimsekdanilo, 2019-04-23:
@xiongyihui @SethWelday Could you please send the config file for the respeaker core v2.0 to me too?

My email is: 
zimsek.danilo@gmail.com

thanks

tianshuo, 2019-05-08:
@zimsekdanilo @xiongyihui @SethWelday Could you please send the config file for the Respeaker Core v2.0 to me too?

My email is tianshuo@gmail.com

Thanks a lot!

MaulikSS, 2019-08-26:
@xiongyihui @FrancoisGrondin @SethWelday What would be a good resource for me to procure the config file for the ReSpeaker Core v2.0 and also understand how to use it along with ODAS? I've checked @xiongyihui's repo mentioned above, and it has a config for the 6-mic array. Can the same config be used with the Core v2.0? Also, @FrancoisGrondin is there any documentation on how to use the odaslive library by itself, without the web interface? Any help would be greatly appreciated :)

nxdefiant, 2019-08-26:
Doesn't the config file from the [pull request](https://github.com/introlab/odas/pull/72/files) work for you?

MaulikSS, 2019-08-28:
@nxdefiant Thanks, that config worked for me.
However I am getting suboptimal sound recordings. I'm still trying to debug that issue, and will post an update here. If you have any suggestions, please let me know. (I've located a similar issue posted here).

Another question @FrancoisGrondin, I am facing the abrupt crashing of odaslive on the ReSpeaker Core v2.0, with ODAS studio being run on my PC. CPU utilization ranges from 130-150% for the odaslive process, and it abruptly exits by itself in about 20-30 seconds. Sometimes however, it sustains itself for 10 minutes, until manually quit. Could you please help with this issue? Any help will be greatly appreciated. Thanks!