introlab/odas #125: odascore

clovett, 2018-10-18:
[https://respeaker.io/4_mic_array/](https://respeaker.io/4_mic_array/) talks about "odascore", but when I build this project I only see odaslive and odasserver in the folder, and odascore is nowhere to be found... have things changed since that wiki was written?

FrancoisGrondin, 2018-10-18:
Hi there,

Yes this has changed, it is now odaslive... We are aware this is confusing, and this is why we are currently working on the first official release that will hopefully solve most issues like this one.

Thank you,

clovett, 2018-10-18:
ah, got it, thanks, and I found the config/odaslive/respeaker_4_mic_array.cfg file.  I edited the server ip to connect to the odas_web app on my PC and bingo, works, seems to be locating sound in 3D, very cool work.
[image]