introlab/odas #130: Running odas_web and a python script at the  same time

viduraakalanka, 2018-11-05:
Hi, currently  i am using odas_web with my Respeaker mic array v2.0. I want to  know whether we can run odas_web and at the same time another program which uses the mic array (i.e. python script). If so, how can  we implement it?

FrancoisGrondin, 2018-11-22:
Hmmm you mean two devices that use the ALSA driver? Technically you'll need to have a script that gets the input from the device, and then pipes it to two pipes, one that goes to odas, the other to your python script.

dianalow, 2020-05-21:
I'm also facing this issue. I got some clues reading scotthasse's [solution for the Delta audio card](https://github.com/introlab/odas/issues/21#issuecomment-376580146)

and tried to configure my /etc/asound.conf to this:
```
pcm.multi_capture{
    type dsnoop
    ipc_key 1024
    slave {
        pcm "hw:1,0"
        channels 6
    }
}

pcm.dmixed {
    type dmix
    ipc_key 1024
    ipc_key_add_uid 0
    slave.pcm "hw:0,0"
}

pcm.duplex {
    type asym
    playback.pcm "dmixed"
    capture.pcm "multi_capture"
}

# Instruct ALSA to use pcm.duplex as the default device
pcm.!default {
    type plug
    slave.pcm "duplex"
}
ctl.!default {
    type hw
    card 0
}
```
I tested the configuration to see if it works:
`arecord -Dplug:multi_capture -f S16_LE -c 6 test.wav`
and it does.

I then modified my odas config to this:
```
raw: 
{

    fS = 16000;
    hopSize = 128;
    nBits = 16;
    nChannels = 6; 

    # Input with raw signal from microphones
    interface: {
        type = "soundcard_name";
        device_name = "multi_capture";
    }

}
```

but when I ran odaslive, I encountered this error:
`Cannot find path "raw.interface.devicename" in config file`

Am I using the `soundcard_name` field incorrectly?

dianalow, 2020-05-22:
My bad - it is "devicename" instead of "device_name". 

I've managed to launch both odaslive and arecord concurrently. Haven't had any luck with my python script but it could be how my script is written.

Hopefully this helps in the solution somehow!

Anamtya2024, 2025-02-03:
Life saver....thanks for this...