introlab/odas #131: ODAS odaslive Seeed ReSpeaker Mic Array v2.0 configuration

kyle-redyeti, 2018-11-21:
Hello,

I have been trying to do a little Thanksgiving Hack-A-Thon and I am trying to follow the tutorials to get my new Mic Arrays up and running with ODAS Studio. It looks like it will be very cool and should help me achieve my end goal of determining the location of a sound or speaker. I am trying to use the respeaker_4_mic_array and it stops at Launch threads..... with an error Sink pots: Cannot connect to server. Is there something I am missing?

Many thanks!

Kyle

FrancoisGrondin, 2018-11-22:
Hi Kyle,

The source hops error means ODAS is not able to open the sound card with the specified sample format. Have you check to make sure the sample format you are using (in bits) is supported by your sound card?

Happy thanksgiving!

kyle-redyeti, 2018-11-22:
I have not I will check to see if I can dig that info up. I am guessing it will be the RaspPi versus the ReSpeaker Mic Array v2.0... I will take a look after my Turkey Coma subsides... 

Thanks!

kyle-redyeti, 2018-11-24:
Hello! Thanks for your help I was able to get everything up and running and it looks like with running ODAS, ODAS Web and ODAS Studio on my little Rasp Pi 3 B+ with Ubuntu Mate it runs ODAS Studio for a very short time before I run out of memory and everything stops. I know I can run ODAS Studio on another machine. Can I also consume the data running from odaslive from another application. Do I need to run ODAS Web and ODAS Live from the RaspPi and then I would have access to the data via the ports? I really would like to be able to feed the data of distance and direction into another application.  Also, second crazy question can I run two ReSpeaker mic array v2.0 and have them processed through ODAS. Would I have to just try to mangle the config file to allow more sources and their orientation? Or would I need to add a second version of ODAS running? I am looking to use a pair of them to do a "full" 360 location of audio source... Any guidance you can give would be appreciated on any of these efforts!

Many Thanks!

Kyle

kyle-redyeti, 2018-12-07:
> Yes, technically you just need to read the port that is normally read by
> You would then need to have both multichannels streams to be concatenated
> as a single stream (maybe using a tool such as sox). But then both sound
> cards will not be in sync, and this will introduce issues regarding
> localization. My suggestion is that you use a single array that spans a 3D
> plane instead, and then you can do full 3D localization.
> Hope that helps!

Thanks for your previous help! I really appreciate your assistance in
getting me past the stumbling blocks I initially hit when beginning to
explore this powerful software.  In my specific project I am attempting to
pull the localization information of the incoming sound but also do speech
to text of the actual audio.  I will have two microphone arrays connected
to a raspberry pi and it looks like I can modify the config file to charge
the orientation of the microphone locations which SHOULD allow me to put
the two sets of data into the same reference frame. Still working on this
but I think I see the path forward.

The second challenge is that I want the audio signal to also be sent off
the Raspberry Pi so that I can process it.  I was looking at something
REALLY basic like rtp but it looks like I can not connect to hw1,0 or hw2,0
because when I connect odaslive it is marking them as busy.

Is there anything that is currently available in odas to stream the audio
out? Or am I completely missing a simple answer???

Many Thanks!

Kyle

EazyRob97, 2019-03-07:
Hi there, I'm working on an interactive Robotic Lamp (LampBots) Project with my team in college, where I have to use a software such as this perhaps to localize sound coming in accross the x-axis and y-axis where it would then take in the coordinates of a sound and send it to our vision software that we use called Roborealm to look at the person that is speaking and that it includes when 2 or more people are in the picture and be able to respond, in order to give it more of a feel!, My job is to localize sound and send its value over to a vision software, would this Software be capable of that?, and if so where can I get started, and where would be able to get a suitable microphone array to work with this? (such as the xmos xcore 7-microphone array that I came accoss while reading through this repository)?, would this work with the software as well? https://www.seeedstudio.com/ReSpeaker-Mic-Array-Far-field-w-7-PDM-Microphone-p-2719.html Thanks!

EazyRob97, 2019-03-07:
Hi there, I'm working on an interactive Robotic Lamp (LampBots) Project with my team in college, where I have to use a software such as this perhaps to localize sound coming in accross the x-axis and y-axis where it would then take in the coordinates of a sound and send it to our vision software that we use called Roborealm to look at the person that is speaking and that it includes when 2 or more people are in the picture and be able to respond, in order to give it more of a feel!, My job is to localize audio sound coming in and send its value over to a vision software, would this Software be capable of that because it looks really cool ?, and if so where can I get started, and where would be able to get a suitable microphone array to work with this? (such as the xmos xcore 7-microphone array that I came accoss while reading through this repository)?, would this work with the software as well? https://www.seeedstudio.com/ReSpeaker-Mic-Array-Far-field-w-7-PDM-Microphone-p-2719.html Thanks!

kyle-redyeti, 2019-03-07:
With the Respeaker Mic Array you can definitely have it provide the
direction of what seems to be the loudest (current) speaker. With a bit of
work you can publish this using some sort of pub/sub interface and then
send api commands to move a ptz camera or run servos to move a camera. You
should be able to do all the processing you need on a RaspPi!

Hope that helps!

EazyRob97, 2019-03-08:
We have a vision system that tracks human faces and gets the x and y coordinates of each face. We also have a PCB board in which we send serial commands to move the lampbots, so if I'm just wondering exactly how I'm going to send in the x,y coordinates from where the loudest energy in this case is coming from and publishing it to our vision systems, you said a RaspPi would be suitable for this process ...what pub/sub interface are you referring to, something like mqtt publisher & subscriber interface ?

kyle-redyeti, 2019-03-08:
Robert yes I was thinking something like MQTT for Pub/Sub. There is a
couple of example code snippets that you can use to display the direction
of arrival. I would recommend wrapping that response with python to then
publish the data.

EazyRob97, 2019-03-10:
Do you mind giving me your contacts in case I have questions on ODAS configuration guide on my respeaker?, because my college have ordered some for me to use for my project, my email is 20079462@mail.wit.ie  ...how will mqtt come to play when my respeaker listens for sound?, I'm not quite understanding that concept because we already have a vision system (roborealm) to manage vision?, which bring thanks!

mitchhh22, 2020-10-12:
Hi @kyle-redyeti I'm looking for something similar with odas (return x,y coordinates of sound location) using a ReSpeaker.

You mention example code that can be wrapped with python to achieve this, can you please provide a link? I'm having a hard time finding it. Thanks!

kyle-redyeti, 2020-10-13:
It has been awhile but I will take a look for that code and get back to you.

hltkts123, 2021-03-04:
> It has been awhile but I will take a look for that code and get back to you.
> […](#)

HI, can you please give me a file? thanks in advance

kyle-redyeti, 2021-03-04:
I actually started looking at this again last week... I am trying to
integrate it with ROS and run it on a Jetson nano.  I will take a look when
I am at my desk today

Kyle

kyle-redyeti, 2021-03-04:
Can you send me an email at kyle<dot>redyeti<at>gmail with what you are
trying to achieve I am not finding my original python code but if you are
using ROS as your "middleware" I can point you to some of the things I am
working on!

Kyle