introlab/odas #132: How to get the azimuth  from SSL result ?

OrangeBaoWang, 2018-11-22:
Hi , I run odas follow  the guid. My micro array is Respeak 4 mics ,I get the result ,but I don't know how to convert this to a azimuth. Can you help me ?  I only want to  use SSL , how can i stop run other modules ?
{
    "timeStamp": 425,
    "src": [
        { "x": 0.924, "y": -0.171, "z": 0.342, "E": 0.484 },
        { "x": 0.992, "y": 0.115, "z": 0.059, "E": 0.345 },
        { "x": 0.786, "y": -0.616, "z": 0.059, "E": 0.142 },
        { "x": 0.647, "y": 0.681, "z": 0.342, "E": 0.072 }
    ]
}
{
    "timeStamp": 426,
    "src": [
        { "x": 0.000, "y": 0.000, "z": 1.000, "E": 0.674 },
        { "x": 0.000, "y": 0.000, "z": 1.000, "E": 0.000 },
        { "x": 0.000, "y": 0.000, "z": 1.000, "E": 0.000 },
        { "x": 0.000, "y": 0.000, "z": 1.000, "E": 0.000 }
    ]
}

jake266637, 2018-11-22:
float elevation = atan (zdir/sqrt(pow(xdir,2) + pow(ydir,2))) * (180/M_PI);

something like this works

closed 2018-11-27 by OrangeBaoWang

hltkts123, 2021-03-03:
> float elevation = atan (zdir/sqrt(pow(xdir,2) + pow(ydir,2))) * (180/M_PI);
> 
> something like this works

Can you explain the formula?