introlab/odas #135: "E" information in tracked output

jake266637, 2018-11-28:
I am wanting to output the "E" value instead of "activity" value in the tracked output (which i made into a raw file), but i'm not sure how to go about this exactly. Can anyone shed some light?

jake266637, 2018-11-29:
Also how is "E" calculated?

FrancoisGrondin, 2019-01-10:
@jake266637 

Sorry for the late reply. I just pushed the paper on ODAS online, and you can access it here: https://arxiv.org/pdf/1812.00115

This should answer many of your questions

jake266637, 2019-01-10:
Great thanks, so is there a way to get the "E" value to show up against the tracked sources instead of the "activity" value which doesnt seem as useful? Or is that not possible at this point in time?

FrancoisGrondin, 2019-01-10:
Oh; for now I did provide the activity, which is the probability the source is active, and considers both the energy (E) and the position (x,y,z). I could eventually add the E if you think this is useful.

jake266637, 2019-01-10:
Hmm i think it could be useful. Or if you could confirm whether the post-filtered audio output is proportional to the input audio? Or if there was a way to access the raw audio before filtering?

Femtofirst, 2023-07-03:
Hi, I am wondering what does 'x' 'y' 'z' and 'e' means in simple words? As a non-technical person the paper is a bit hard to read for me....