introlab/odas #136: Stretching of audio files

jake266637, 2018-11-30:
Hey i have just realised when comparing a unix time stamp to the "time stamp" (which is number of timestamps times 0.008 seconds) that the ODAS recording stretches the audio by a factor of approximately 1.008. I recorded a sound in odas and compared this to the actual audio file in audacity and confirmed the stretching effect.

Could this be due to editing the ODAS code from the stock version?

nxdefiant, 2019-08-09:
I did not fully understand your post, but I think I have a similar problem. I was running odaslive with separated raw sss output. In about* one hour the .raw file increased by 464515072 bytes.
With 16 Bits, 4 channels and a rate of 16kHz after one hour I expect the file size to be 2*4*16000*60*60=460800000 bytes.

That is a difference of 464515072/460800000 = ~1.008

* Manually measured with only an accuracy of one second..

FrancoisGrondin, 2019-08-09:
I'm not sure I understand... 

@jake266637 What exactly do you mean by recording a sound in odas?
@nxdefiant What do you mean exactly by "running odaslive with separated raw sss output"?

nxdefiant, 2019-08-09:
I meant this in my odaslive config:
```
sss:
{
    separated: {
        interface: {
            type = "file";
            path = "/tmp/separated.raw";
        }
    };
}
```

jake266637, 2019-08-15:
> I'm not sure I understand...
> 
> @jake266637 What exactly do you mean by recording a sound in odas?
> @nxdefiant What do you mean exactly by "running odaslive with separated raw sss output"?

I believe I made a sound file in audacity that had impulse sounds in it, ODAS said the pulses were further apart than they actually were. This was nearly a year ago now so I can't exactly remember but I do know that the stretching effect was apparent and equal to approximately 1.008 which is the same value nxdefiant found.