introlab/odas #142: ReSpeaker 4 mic Linear Array socket connections

PabsBraz, 2019-01-15:
Hi guys,

First, thanks for any given help. I am currently starting to work with a ReSpeaker 4 mic Linear array. Following the tutorials, I installed the ODAS library from the Phoenix branch and ODAS studio. 

I am running ODAS studio on a PC with Ubuntu 16.04 and a Pi 3 with Raspban Stretchp running ODAS Core. I am able to record with the array and it shows up when listing with arecord -l.
However, I am not able to represent data on ODAS studio by trying to adapt the config files with my PC's IP. Therefore, I have got to the following conclusions and doubts:

1. Firstly, I take into account that the config file found for the ReSpeaker for the 4 mic array is not applicable to the linear array due to geometry. However, would this be a possible problem for not receiving data? I thought it would just affect calculations and it is something I take into account for further development, having also checked #18. 

2. I have noticed that after building, the executable is called **odas**, and the config file is divided in Config and Socket JSON files. I have tried putting the IP address in the socket file as follows (must admit that following my intuition and I am quite sure this is my main error) and was able to start processing. However, I still wasn't able to represent any data. This is what I added to *ssl.json*:
```json
{
    "sockets":
    {
        "raw": 5001,
        "targets": 0,
        
        "pots": 9001,
        "tracks": 0,
        "seps": 0,
        "pfs": 0,
        "iseps": 0,
        "ipfs": "pc_ip_address"
    }
}
```

3. To see if it's a problem with the Server, I introduce my IP and port in the raspberry browser, and I can observe that the studio detects the connection. 

So, after a couple of hours trying different things out and diving into the issues section, I believe the problem is in the *socket.json* and *config.json* configurations, and therefore I ask... Which is the correct way to write a configuration file with this two-file method? :nerd_face: 

Thanks in advance once again and I look forward to your replies!

FrancoisGrondin, 2019-01-17:
Hi,

So a quick answer is that for now unfortunately the ODAS Studio is only compatible with the ODAS in the master branch :P Did you have the chance to play with the version on the master branch as well?

Cheers,

PabsBraz, 2019-01-17:
Thank you very much for the reply! 

That's good to know, I'll suggest an edit of the tutorial [here]( https://github.com/respeaker/mic_array#use-4-mic-array-with-odas-to-sound-source-localization-and-tracking ) in that case.

I did install master, but ran into some issues with my dependencies and, must admit, got lost into code. Therefore, I'll get right to it and test it out! 

About the configuration file, should I open another thread to comment about the possibility of a *4_mic_linear.cfg* file? I thought of starting by basing myself on the *4_mic_array.cfg* there is right now and change the coordinates of the mics, plus check your doc on the PSeye3. 

Let me know what you think about this ;) 

Cheers!

PabsBraz, 2019-01-18:
It's working! Using the master branch and connecting remotely already. 

However, just in case it hasn't been mentioned, I am not able to finish the odaslive process by keyboard interruption. I have to kill the process after looking for its PID. 

I am now tweaking the configuration file to correctly represent the data. Therefore, I'll wait for you to tell me if you would like me to close this.

Thank you very much for the help :+1:

FrancoisGrondin, 2019-01-19:
Great! Happy to see you got it working! Yes I'm good about closing the issue.

Cheers

closed 2019-01-19 by FrancoisGrondin