introlab/odas #143: input wave file

bright357122, 2019-01-17:
Hi IntRoLab,
We would like to run the ODAS without actual microphone array, instead, by loading recorded wave files from each mic on the array.
But we can't find the entry or API to load the files. Could you give us a hint of this?

bright357122

FrancoisGrondin, 2019-01-17:
Hi,

Have a look here: https://github.com/introlab/odas/wiki/Configuration#interfaces

You can actually specify a RAW file (if you only have wav files, you can convert them to the RAW format using a tool such as Audacity or Sox)

Cheers,

dangpnh, 2019-07-12:
@francoisferland 
Sorry in case there're multiple files, how do I set the raw configuration to receive all of them?

francoisferland, 2019-07-12:
Hi @dangpnh, I am not very familiar with ODAS, but you will get better information from @FrancoisGrondin.

Junye97, 2019-09-13:
@dangpnh 
I got around this problem by converting multiple single track .raw files into a single multi-track .raw file using MATLAB (since audio recording softwares like Audacity does not support exporting multi-track .raw files.... strange). I'm sure there must be much simpler ways so please let me know if you figure out other solutions...

materialvision, 2022-05-20:
Any hints to how you did this in Matlab? 

> @dangpnh
> I got around this problem by converting multiple single track .raw files into a single multi-track .raw file using MATLAB (since audio recording softwares like Audacity does not support [...]

liuyoulong0731, 2025-05-14:
> [@dangpnh](https://github.com/dangpnh) 我通过使用 MATLAB 将多个单轨 .raw 文件转换为一个多轨 .raw 文件解决了这个问题（因为像 Audacity 这样的录音软件不支持导出多轨 .raw 文件……真奇怪）。我相信肯定有更简单的方法，所以如果你找到了其他解决方案，请告诉我……

Using adobe audition, he can create a multi-track file with any number of channels, and easily copy multiple single-track files into the multi-track file