introlab/odas #145: Official Release Date

janzuiderveld, 2019-02-06:
How is the progress of the official release doing? Could you give an approximation of when it (and the complete tutorial) will be released?

FrancoisGrondin, 2019-02-07:
It is still in progress; hard to give an exact date when it will be ready as I'm working on this in my spare time. Regarding the tutorial, there is however now a paper available that explains all the technical justification behind the localization+tracking in ODAS: https://www.sciencedirect.com/science/article/pii/S0921889017309399

raikarsagar, 2019-08-29:
Hi, When can we expect the new version to be released? Also could you please provide any reference for the parameters defined in config file.(description about each parameter)

FrancoisGrondin, 2019-08-29:
We don't have a release date yet. There is a wiki however that explains the parameters:

https://github.com/introlab/odas/wiki/Configuration