introlab/odas #156: result too bad

11721206, 2019-05-23:
hello, thanks for this great projects.
I configure my projects files as follows:
[my_respeaker_4_mic_array.txt](https://github.com/introlab/odas/files/3211400/my_respeaker_4_mic_array.txt)

and my microphone array is put as follows:
[image]

but I get the result two bad.
my source is a frequency = 1000hz sine wave, the location is (0.1, 0 , 0.05)(cm).
but my result is too bad as follows.
[mic3.log](https://github.com/introlab/odas/files/3211405/mic3.log)

the test pcm is:

[mic3.txt](https://github.com/introlab/odas/files/3211410/mic3.txt)
 for uploading ,I changed mic3.pcm into mic3.txt.

any suggestions? thanks very much.

FrancoisGrondin, 2019-05-25:
Hi there,

The reason why the system is not doing well is because you have a narrowband signal (a 1000 Hz sine wave), and ODAS relies on GCC-PHAT, which works only for broadband signals such as speech. Unless the 1000 Hz signal is your real target, I suggest you use a wideband white noise signal to test your system, which should work very well.

11721206, 2019-05-26:
> Hi there,
>
> The reason why the system is not doing well is because you have a narrowband signal (a 1000 Hz sine wave), and ODAS relies on GCC-PHAT, which works only for broadband signals such as [...]

thank you for your reply

closed 2019-07-10 by FrancoisGrondin