introlab/odas #157: is this project can beamforming?

11721206, 2019-05-26:
sorry to bother you again.
I want to know whether this project can beamforming? I see some infomation assoicated with beamforming in steer2demixing.c.
the configuration file contains 'dmvdr', but nothing in the item.
and the project in manyears-C contains Beamforming function but is seperated from this project.
 if I want to beamform the channel signals after sound source location, is there any suggestion for my work on odas? thanks!

FrancoisGrondin, 2019-05-27:
Yes you can do beamforming with the separation module: it supports delay and sum beamforming, and geometric source separation beamforming. MVDR could be added but wasn't added yet as it would take much more computations, and make real-time difficult on low-cost hardware. Everything is in place though to code it in the future.

StuartIanNaylor, 2022-02-22:
@FrancoisGrondin do you think it could be possible on a Pi4?