introlab/odas #158: Can I record the sound  only  in fixed direction?

Nonikka, 2019-05-29:
hi, I have a usb 4 mic array(ReSpeaker_Mic_Array_v2.0). 
The situation is: I want record(enhance) the voice in the range 0°~10° or 0°~90° or a specific direction, even there is no sound.
I have try the postfiltered sss,it is good in some situation, but it is dynamic  tracking the sound, instead of one specific direction.
Could you give me some hint? (like change the .cfg file or how to modify the code like force give a x,y,z as a src for separation the voice)
Thanks!

Nonikka, 2019-05-29:
[image]

FrancoisGrondin, 2019-05-29:
Hey there!

Yes you can do this:

First set this flag in the config file in the sst block:

```
# Add is either "static" or "dynamic"
add = "static";
```

And then set the following variable, where (x,y,z) should take the value of the normalized direction of your target (e.g. it could be say x = 1.0, y=0.0, and z=0.0):

```
target: 
(
        { tag = "myTarget"; x = 0.0; y = 0.0; z = 1.0 }
);
```

Hope this helps!

Nonikka, 2019-05-30:
> Hey there!
>
> Yes you can do this:
>
> First set this flag in the config file in the sst block:
>
> ```
> # Add is either "static" or "dynamic"
> add = "static";
> ```
>
> And then set the following variable, where [...]

This is very useful！ I have another question, what does the "ss","ms" mean in the "post-filtered", and also the "dds","dgss" in "separated"? 
Thanks again.

closed 2019-06-03 by Nonikka

Nonikka, 2019-06-03:
> Maybe I can help you. The SSS separation methods have three mode.dds,dgss and dmvdr.In that,the ds is just the delay-and-sum,the dgss is Geometric Source Separation,and the mvdr is not to be done in the code.
Actually, I think that the most useful to the separation is the post-filter.The post_filter have two ways,ss and ms.the ss is one channel and ms is mutli-channel.the ms is much better than ss.In the ms,the odas use mcra and reduce the leakage from other output channel.And more,it gives a suppression rule in teh presence of speech.

anrzej96, 2020-03-23:
I think like I have similar issue. 
I'm desperately looking for help. Could you help me?
https://github.com/introlab/odas/issues/194

pfeatherstone, 2020-08-24:
@FrancoisGrondin Can you also achieve spatial filtering using:
```
spatialfilters = (

        {
            direction = ( +0.000, +0.000, +1.000 );
            angle = (80.0, 100.0);

        }    

    );
```
??
I also want to spatially filter sound given an azimuth and elevation. I can't get it to work using either method.

pfeatherstone, 2020-08-24:
Also, is there a way of using the C API rather than using the configuration file.