introlab/odas #159: Mapping program and channel streams

seanernestmurray, 2019-06-12:
Hello!

I'm currently tracking up to 4 sources, using an 8 mic configuration of the 16 sounds USB and am looking to obtain the tracked data and separated audio of each source together in real-time. If I'm tracking 4 sources, should I expect 4 channel streams in the raw separated/post-filtered audio?  

Issue #79 indicates that there is a direct mapping of the SST data and the channel streams based on order (e.g. if the second line is tracking a source, this corresponds to the second channel in the raw audio). However, #87 deals with separating the raw audio into specific channels, i.e. the use of the mapping program from the phoenix branch. If the channels from the separated and post-filtered raw audio already correlate to the SST data, then what is the function the mapping  program?

For example: In my 8 mic configuration, the order of tracked sources in the SST data appears to correctly match the channel order in the separation/post-filtered streams when I import the audio (at 16-bit PCM) into Audacity with 4-channels. However then, what is the use of the mapping program once I have this data?

PS: what is the function of the sequencer tool in the same folder?

Thanks!

seanernestmurray, 2019-06-12:
Also, is the tool that can write each separated source to a file with the corresponding id in the name, referenced in #79, available?

FrancoisGrondin, 2019-06-26:
Hi there!

Sorry for the late reply. I have a few questions:

1) Are you using the ODAS version in the master branch?
2) If I understand properly, your goal is to write each separated stream to a file with the corresponding ID?

Best,

seanernestmurray, 2019-06-27:
Many thanks for your reply!

1.
Yes, I'm currently using the ODAS version in the master branch. On the sink end, I've edited a program from the tools directory of the phoenix branch in order to receive the sst data on port 9000 and publish it onto a ROS topic, which seems to be working. The idea is to have both the tracking data and separated audio easily accessible for each source, on ROS.

2.
In fact, the current goal is to publish each separated stream to a topic, with its associated source ID, using a ROS stack such as _audio_common_ (I've given a sample msg type below). I figured this would be relatively straightforward if I had access to the tool you referenced in the end of your answer to #79, that which writes each separated stream to its own file with corresponding ID. 

Of course, if you have suggestions of an alternative or more optimal approach, that would be very welcome also!

AudioData.msg sample: (up to 4 sources)
```
uint32[] src_ids
uint8[] data1
uint8[] data2
uint8[] data3
uint8[] data4
```

Thanks again,

FrancoisGrondin, 2019-06-29:
Oh ok! So you have the stream but what you are missing is a way to push it to a wav file, correct? I suppose the problem now is that you'd like to generate the proper headers for the wave file?

seanernestmurray, 2019-07-01:
No, not quite. The goal is to not to push the channels to a wav file - instead, we'd like to separate the channels and stream them in real-time. Do you know if there is an efficient way to do this?
I was thinking that the best way might be to augment the code file you referred to in #79, since it would seem that in this program you are exposing each of the separated channels.

FrancoisGrondin, 2019-07-08:
Hey there, sorry for the delay.

I will try to write a simple code here that hopefully can get you going to fix this issue:

1) As you know, you have a track signal with the following structure:

```
unsigned int nTracks;
unsigned long long * ids;
char ** tags;
float * array;
float * activity;
```
The `ids` array is made of `nTracks` unsigned long long elements. This is what matters here for our code.

2) Then, the hop signal has the following structure:

```
unsigned int nSignals; 
unsigned int hopSize;
float ** array;
```
The array `array` is made of `nSignals` float * arrays, and each float * array is made of `hopSize` float samples. For separated signals, odas makes sure `nSignals = nTracks`.

3) We then have the following correspondance: suppose we have a maximum 4 tracked/separated sources, and two currently tracked/separated sources. The positions of the IDs in the list of ids are assigned randomly, because it depends on the available slots when a new source is created. For instance, if we have two sources with ids 5 and 9, we could have:

```
ids[0] = 9;
ids[1] = 0;
ids[2] = 5;
ids[3] = 0;
```
Note how the ids are positioned randomly, and the free slots are set to an id of 0. So, with this configuration, it means we have:

```
array[0] = <array of `hopSize` floats, which samples correspond to the current frame in the time domain for the tracked source id = 9.
array[1] = <array of `hopSize` floats, set to 0's everywhere>
array[2] = <array of `hopSize` floats, which samples correspond to the current frame in the time domain for the tracked source id = 5.
array[3] = <array of `hopSize` floats, set to 0's everywhere>
```

So then the goal is to copy the samples from source id = 9 to a given stream, and source id = 5 to another stream.

4) For simplicity, we will assume we want to write the separated streams to files. So we will keep a list of file pointers:

`files = <array of `nTracks = nSignals` FILE *>`

By default all FILE pointers are set to NULL.

There would be 3 steps. Note that here we assume that when a source disappear, its ID is set to 0, and when a source appear, a ID previously set to 0 is set to the new ID value. We assume a source cannot jump from a given ID to a new one (if that source is deleted and a new one is created at the same time).

A) Open new files for newly created source

The pseudo-code would look for each IDs, and when an ID is not 0, it would check if the file pointer at the same position is not Null. If it's Null, then you open a file (using fopen) and assign this file pointer to the corresponding slot.

Ex: We initially only have a source with Id = 5, and then Id = 9 appears. We thus have:

```
ids[0] = 9; files[0] = NULL;
ids[1] = 0; files[1] = NULL;
ids[2] = 5; files[2] = <Pointer to opened file 1>;
ids[3] = 0; files[3] = NULL;
```

and this becomes:

```
ids[0] = 9; files[0] = <Pointer to opened file 2>;
ids[1] = 0; files[1] = NULL;
ids[2] = 5; files[2] = <Pointer to opened file 1>;
ids[3] = 0; files[3] = NULL;
```

B) Push data in their corresponding file

Then we loop in the hops signal, and push each stream to the corresponding opened file:

```
samples in array[0] copied to <file 2>;
samples in array[2] copied to <file 1>;
```

C) Close files for deleted source 

Finally, we checked if any source was deleted, to close the corresponding file. Let's suppose that a few frames later the source 5 disappear. We would then have the following:

```
ids[0] = 9; files[0] = <Pointer to opened file 2>;
ids[1] = 0; files[1] = NULL;
ids[2] = 0; files[2] = <Pointer to opened file 1>;
ids[3] = 0; files[3] = NULL;
```

We see that we have a 0 id at position 2, with a non-null file pointer. We thus close the corresponding file and reset the pointer to null:

```
fclose(<Pointer to opened file 1>);
ids[0] = 9; files[0] = <Pointer to opened file 2>;
ids[1] = 0; files[1] = NULL;
ids[2] = 0; files[2] = NULL;
ids[3] = 0; files[3] = NULL;
```

That's pretty much it. Is this what you are trying to achieve? With this example you would get streamed written to files in RAW format. The easiest way to proceed is to name the files with their corresponding ID. You can also do a similar thing with sockets for instance.

Cheers,

seanernestmurray, 2019-07-09:
Hi, 
Many thanks for your thorough reply! 

The only thing I am still uncertain on is how to get access to the track/hop variables (ids, float ** array, etc...) in the first place? Should I parse the track/hop signal data from the sss sockets, create a src_hop_obj, or are they accessible elsewhere? 

All else seems to make sense. Thanks again.

FrancoisGrondin, 2019-07-09:
You have two options: you can either access the track/hop directly within the C code, or you can send the track message as a JSON via the socket, and the separated stream as a RAW signal via the socket. If you decide to proceed via sockets, then you'll have to do some parsing (can be done with outside ODAS then) at the other end to extract the proper information.

seanernestmurray, 2019-07-09:
I am now correctly parsing the track message via a socket, so that isn't an issue. 

I am unclear on parsing of the RAW separated stream signal. In each frame that is received through that socket, do the first 2 bytes correspond to the _unsigned int nSignals_, the next 2 bytes correspond to the _unsigned int hopSize_, and so on? 

Once those values are parsed, the rest seems clear to me.

FrancoisGrondin, 2019-07-09:
No, so in the case of the RAW signals, you get only the samples, which are interleaved, which corresponds to a typical RAW format that you could use with Audacity for instance. You can infer `nSignals` from the track message `nTracks` parameter, which corresponds to the number of slots.

If you are outputting the RAW in 16 bits, then the function `snk_hops_process_format_binary_int16` gets called to serialize your samples, and, with say nTracks = nSignals = 4, you'll get something like:

```
<Ch 1 - Sample 1, 16 bits >
<Ch 2 - Sample 1, 16 bits >
<Ch 3 - Sample 1, 16 bits >
<Ch 4 - Sample 1, 16 bits >
<Ch 1 - Sample 2, 16 bits >
<Ch 2 - Sample 2, 16 bits >
<Ch 3 - Sample 2, 16 bits >
<Ch 4 - Sample 2, 16 bits >
...
```

seanernestmurray, 2019-07-11:
Ah of course, that's all clear now. Many thanks for all your help :) 
Cheers,

FrancoisGrondin, 2019-07-11:
Glad it helped!

Cheers

closed 2019-07-11 by FrancoisGrondin

TankyFranky, 2023-05-19:
I would like to extend this question by asking what happens when two sources are added simultaneously? How do we know which audio channel is associated with which ID?

In the situation where ODAS is started with two speakers talking simultaneously, are there defined criteria such as "the lower ID is assigned the lower position in tracked ID list" ?