introlab/odas #164: Error in "npm install"

deepchand41, 2019-07-04:
Hello @GodCed 
After cloning the odas_web, I am getting error as shown in below figure while running "npm install " in the base folder.
[image]

My npm version is 6.9.0
and node version is v12.5.0
It will be a great help if you reply. 
Thanks in advance!!!

GodCed, 2019-07-04:
Hi, I'm not sure what's going on here. You seem to have all the proper versions... From what I see the grpc module failed to build for electron which is unexpected with the correct versions installed.

I changed the way it is installed in the package.json file (from installing with default node runtime then rebuilding for electron to directly installing the proper binary for electron). Can you try checking out the `grpc-native-install`branch and retry the install?

MohamedSamiGamalElden, 2019-07-13:
try to install node10.x on your OS. It works with me.
https://github.com/nodesource/distributions/blob/master/README.md

abhinit310, 2021-05-27:
Hi
I have tried with node 10.x . But the error remains the same.
[image]

abhinit310, 2021-05-27:
@MohamedSamiGamalElden @GodCed @deepchand41 
https://github.com/introlab/odas/issues/164#issuecomment-849252136