introlab/odas #165: changing the tracking behavior using the config

andycavatorta, 2019-07-10:
Thank you for creating and sharing this impressive code!
Where can I find more information about the config values for SST and SSS?  
I am using ODAS to locate relatively static human voices that are close together.  ODAS is tending to create one source and slowly track it from person to person.  I need to make ODAS either track one source quickly or create separate/new sources for each person.  I don't need the source IDS, just the location data.

 - thanks

FrancoisGrondin, 2019-07-10:
Hi Andy,

So what you may want to do, provided that you're using the Kalman filter, is to reduce the value of sigma_Q^2, which corresponds to the variance of the velocity state. If this value is very small (or zero), then the source cannot really move over time, and this should prevent switching from one source to the other.

In the config file, you can simply change the following value (I think the default value is 0.001, so try reducing it):

 ```
kalman: {

        sigmaQ = 0.001;
        
    };
```

Cheers,

Francois

andycavatorta, 2019-07-10:
That did it.  Actually, increasing the value produced the result I wanted.  Thanks!

FrancoisGrondin, 2019-07-11:
Ok great! I'm still a bit confused why increasing the value works... I guess it really depends on the dynamics of your system