introlab/odas #166: using "interface" in matrix_voice.cfg

annetteb3, 2019-07-10:
hi all,
noticed there was no information on interface configuration in the wiki and i looked through several of the threads to see if anybody was having the trouble i am; I am trying to get the audio that odas is processing to another device via gstreamer. I have the device's ip and port that it's listening through but no luck....

I was wondering if there's some steps i need to take to stream that audio through gstreamer. I have not installed odas_web as the rpi3 cannot handle and and the main device (jetson tx2) cannot use the version of ubuntu we need for nodejs v12 because it needs to stay the same for other components of our system.

Any help??

FrancoisGrondin, 2019-07-11:
Hi there,

I'm not familiar with gstreamer: can you provide the framework with RAW audio data?

annetteb3, 2019-07-12:
I'm not entirely sure what is meant by tha question (I also am in a bit over my head) here is a link with more details. The short version however we use a UDP connection to stream raw audio between our two devices.

annetteb3, 2019-07-14:
would i need to open another port in the matrix-odas cpp file?

annetteb3, 2019-07-14:
Actually you can forget my gstreamer question......the audio is too degraded...but i am still wondering how to stream the audio from sss in real time. I sent the audio to files and it was a horrible long beep and static.

nxdefiant, 2019-08-04:
Sounds more like you got number of channels, audio format or the rate wrong. For the record in the sss configuration one can use interface type=socket. Gstreamer should be able to receive that with the tcpserversrc element.

This is the pipeline that I just successful tested with (it will only play the first channel/first separated sound source):
`gst-launch-1.0 tcpserversrc host="0.0.0.0" port=10001 ! "audio/x-raw,format=S16LE,rate=16000,channels=4,channel-mask=(bitmask)0xf" ! deinterleave name=d d.src_0 ! queue ! audioconvert ! autoaudiosink`