introlab/odas #167: Inaccurate tracking above 65 degree elevation

jake266637, 2019-08-01:
When performing some accuracy testing on the matrix creator using the ODAS software I noticed that above 65 degrees the software latched onto this very quiet noise (potentially not from a physical source). This sound isn't loud enough to trigger the tracking by itself but can cause the software to continue tracking in that direction for a long time after the initial sound. When the software is latched onto this sound it inaccurately calculates the direction of arrival of the sources of interest. Increasing the active and inactive threshold values seemed to allow this problem to be mostly ignored but this then limits the usefulness of the software. Do you have any ideas for what is causing this problem?

I am using the software to determine the DoA of individual dog barks occurring some distance away from the monitor where I want each individual dog bark to have it's own ID tag so the calculated direction is accurate.

FrancoisGrondin, 2019-08-01:
This is strange. Do you think you could provide us with some RAW recordings and the config file you are using?

jake266637, 2019-08-01:
[raw audio.zip](https://github.com/introlab/odas/files/3459030/raw.audio.zip)

active mu was between 0.1-0.35 and inactive mu was usually 0.05 less than that. N_inactive is at 15.

jake266637, 2019-08-19:
Here is the config file. @FrancoisGrondin 
# Configuration file for MATRIX Creator Raspberry Pi shield

version = "2.1";

# Raw

raw:
{

    fS = 48000;
    hopSize = 512;
    nBits = 16;
    nChannels = 8;

    # Input with raw signal from microphones
    interface: {
        type = "soundcard";
        card = 1;
        device = 0;
    }

}

# Mapping

mapping:
{

    map: (1, 2, 3, 4, 5, 6, 7, 8);

}

# General

general:
{

    epsilon = 1E-20;

    size:
    {
        hopSize = 128;
        frameSize = 256;
    };

    samplerate:
    {
        mu = 16000;
        sigma2 = 0.01;
    };

    speedofsound:
    {
        mu = 343.0;
        sigma2 = 25.0;
    };

    mics = (

        # Microphone 1
        {
            mu = ( +0.020091, -0.048504, +0.000 );
            sigma2 = ( +0.000, +0.000, +0.000, +0.000, +0.000, +0.000, +0.000, +0.000, +0.000 );
            direction = ( +0.000, +0.000, +1.000 );
            angle = ( 80.0, 100.0 );
        },

        # Microphone 2
        {
[... 286 lines cut ...]

classify:
{

    frameSize = 4096;
    winSize = 3;
    tauMin = 88;
    tauMax = 551;
    deltaTauMax = 20;
    alpha = 0.3;
    gamma = 0.05;
    phiMin = 0.5;
    r0 = 0.2;

    category: {

        format = "undefined";

        interface: {
            type = "blackhole";
        }

    }

}

FrancoisGrondin, 2019-09-14:
I notices recently some hardware have improper phase calibration between their channels (maybe due to the way the codec samples turn by turn each input). This might explain this. I'll try to make a calibration tool to fix that in the near future. Cheers.

jake266637, 2019-11-18:
Hello @FrancoisGrondin 

I was wondering if there was any progress on this issue yet and also if you could suggest some parameters I could change to improve the localisation of more impulsive sounds in a noisy environment (dog barks outside). I find that the DOA accuracy is only accurate if the dog bark is the sound that actually triggers the recording on one of the 4 channels.