introlab/odas #169: respeaker seeed

bartengine27, 2019-08-10:
Dear,

  I managed to install and configure odas and odas-web. After enabling "separated audio", I get audio files like 
[ODAS_140_2019-8-10_23-27-9-79_pf.zip](https://github.com/introlab/odas/files/3489304/ODAS_140_2019-8-10_23-27-9-79_pf.zip)
  Any idea what might go wrong?

Thanks,
Bart

FrancoisGrondin, 2019-09-14:
Sounds like you signal is clipping because the gain is too high. Try reducing the gain and tell me if it helps.

bartengine27, 2019-11-28:
Thanks for the feedback!

Turned out to be the "Sample Rate" setting in ODAS Studio under "Configure". After changing the default  16000 to 44100 the audio improved.

My config:
`
    separated: {

        fS = 44100;
        hopSize = 512;
        nBits = 16;        

        interface: {
             type = "socket"; ip = "192.168.1.198"; port = 10000;
	     #import in audacity as raw data with
	     #signed 16 bit pcm
	     #little endian
             #4 channels
	     #freq 44100
             #type = "file"; path = "separated.raw";
        }        

    };

    postfiltered: {

        fS = 44100;
        hopSize = 512;
        nBits = 16;        

        interface: {
            type = "socket"; ip = "192.168.1.198"; port = 10010;
            #type = "file"; path = "postfiltered.raw";
        }        

    };
`

closed 2019-11-28 by bartengine27