introlab/odas #170: Processing audio in ODAS

Mopele, 2019-08-14:
Hey, 
I would like to kind of use ODAS to get the DOA for a wav file in an iterval of 0.5sec. Would that be possible with ODAS. Or is there a way to save the data from the graphs to use them later on in, for example a machine learning project? 
Kindly, 
Moritz Pfennig. 
PS: I am really a noob...

bartengine27, 2019-08-14:
yes, you can, check the config of the SSS and record sections on https://github.com/introlab/odas_web

Mopele, 2019-08-14:
> yes, you can, check the config of the SSS and record sections on https://github.com/introlab/odas_web

Ok, but I am not familiar with js so is there maybe a way to implement it into python? If not, is there a more in depth wiki of the odas project, because as i see it there is ne real explanation of how to accualy save some data from the analytics right?

Btw thanks for the fast response @bartengine27

closed 2019-08-25 by Mopele