introlab/odas #171: starting odas Error Sink pots

Mopele, 2019-08-18:
Hey, I try to run ODAS like this: .`/odaslive -c ../config/odaslive/respeaker_4_mic_array.cfg`.
My error looks like this: `+--------------------------------------------+
|    ODAS (Open embeddeD Audition System)    |
+--------------------------------------------+
| Author:  Francois Grondin                  |
| Email:   francois.grondin2@usherbrooke.ca  |
| Website: introlab.3it.usherbrooke.ca       |
| Version: 1.0                               |
+--------------------------------------------+
| + Initializing configurations...... [Done] |
| + Initializing objects............. [Done] |
| + Launch threads................... Sink pots: Cannot connect to server
`
If I look at the config everything looks like it should. Is there a way to make it work?

closed 2019-08-25 by Mopele

ajaygunalan, 2019-11-04:
Hey,

I'm also having the same issue. How did you solve it?

Mopele, 2019-11-04:
> 
> 
> Hey,
> 
> I'm also having the same issue. How did you solve it?

You have to start the odas live first. So you open the odas GUI interface and then start the odas client. The explanation is, that you have to have the odas server running to connet to it via the sockets of the odaslive command. If you dont want to use a odas server then have a look in te odasconfig for the respeaker. If you have any more questions just ask ;-)

jiguansghen, 2020-05-16:
?

guelermus, 2022-09-29:
Hi,

I saw an interface called Terminal. 
Would it be possible to use it with odaslive without a need for odasserver?

Best,