introlab/odas #180: SST socket output

MauroPfister, 2019-10-07:
Hi, 

I wrote a little python script with a socket server to receive the SST data streamed by odaslive via a socket client. During some tests I noticed that with a buffer of 2048 bytes (actually down to  1,567) I receive a string containing 4 times timestamp + src data as shown below.  When I lower the buffer size I only receive junks of the SST data but never a complete json with timestamp + src data. 

Hence I was wondering if that behaviour (4 times timestamp + src data) is intended or if I made a mistake in my program? And if so, why do you send the data grouped together instead of just sending one timestamp + src data at a time?

Thanks a lot for your help.

``` json
{
    "timeStamp": 174,
    "src": [
        { "id": 0, "tag": "", "x": 0.000, "y": 0.000, "z": 0.000, "activity": 0.000 },
        { "id": 0, "tag": "", "x": 0.000, "y": 0.000, "z": 0.000, "activity": 0.000 },
        { "id": 0, "tag": "", "x": 0.000, "y": 0.000, "z": 0.000, "activity": 0.000 },
        { "id": 0, "tag": "", "x": 0.000, "y": 0.000, "z": 0.000, "activity": 0.000 }
    ]
}
{
    "timeStamp": 175,
    "src": [
        { "id": 0, "tag": "", "x": 0.000, "y": 0.000, "z": 0.000, "activity": 0.000 },
        { "id": 0, "tag": "", "x": 0.000, "y": 0.000, "z": 0.000, "activity": 0.000 },
        { "id": 0, "tag": "", "x": 0.000, "y": 0.000, "z": 0.000, "activity": 0.000 },
        { "id": 0, "tag": "", "x": 0.000, "y": 0.000, "z": 0.000, "activity": 0.000 }
    ]
}
{
    "timeStamp": 176,
    "src": [
        { "id": 0, "tag": "", "x": 0.000, "y": 0.000, "z": 0.000, "activity": 0.000 },
        { "id": 0, "tag": "", "x": 0.000, "y": 0.000, "z": 0.000, "activity": 0.000 },
        { "id": 0, "tag": "", "x": 0.000, "y": 0.000, "z": 0.000, "activity": 0.000 },
        { "id": 0, "tag": "", "x": 0.000, "y": 0.000, "z": 0.000, "activity": 0.000 }
    ]
}
{
    "timeStamp": 177,
    "src": [
        { "id": 0, "tag": "", "x": 0.000, "y": 0.000, "z": 0.000, "activity": 0.000 },
        { "id": 0, "tag": "", "x": 0.000, "y": 0.000, "z": 0.000, "activity": 0.000 },
        { "id": 0, "tag": "", "x": 0.000, "y": 0.000, "z": 0.000, "activity": 0.000 },
        { "id": 0, "tag": "", "x": 0.000, "y": 0.000, "z": 0.000, "activity": 0.000 }
    ]
}
```

FrancoisGrondin, 2019-10-11:
Typically when you send multiple messages back to back to a socket, you can't guarantee they will be received individually when you read the socket.

pcub, 2019-10-12:
Hello, Im using input.raw instead of sound card in odas. However, when I run odaslive then tracks.txt file was generated. However, all x,y and z values are zero as shown above sample. Does this mean odas did not detected source direction?

Junye97, 2019-10-15:
@pcub I'm not sure about the sst output since I am also trying to figure it out, but I can spot some other issues with your config file: 
1. Most of your output interface are "blackhole" so I reckon you wouldn't see anything at all.
2. For the input file path, I would suggest you put the full file path, e.g. `/home/user/odas/bin/input.raw`, and the same for the outputs. If you don't put the full path odas_web cannot read the input (and it does not spit out errors for this).

pcub, 2019-10-15:
Thanks Junye97, I will test that.

Junye97, 2019-10-15:
@pcub please share your findings with the SST output.... I am still investigating what went wrong with my config. Thank you

MauroPfister, 2019-10-16:
> Typically when you send multiple messages back to back to a socket, you can't guarantee they will be received individually when you read the socket.

Yes, I figured that out while reading up on TCP sockets. I ended up rewriting the javascript code of odas_web in python to parse the incoming json data. The parsing function can be found at https://github.com/introlab/odas_web/blob/66fda0dae550e87afe065e2eeb21152ea6ea1ddc/servers.js#L31-L71. I thought this might be useful to others.

marcus905, 2020-02-26:
Sorry for the off-topic post, but I'm having some of the same issues and I can't seem to find a good source of documentation on how odas works, except reimplementing odas_web.

@MauroPfister by any chance is your python code that decodes the odas output available to read or open source in any way?

Thank you!

Apolotary, 2020-03-24:
@marcus905 did you manage to get any info? Just ran into the same problem myself, would like to use ODAS in my project, but can't figure out how the info is being sent and in what format

marcus905, 2020-03-24:
Not yet, I'm still working on it, but I'll let you know

ameliefroessl, 2020-10-12:
@MauroPfister  I would also be interested in seeing how you receive the data in a python script. I am able to connect to the port which is outputting the SST data, however it seems like no data is being streamed. However, the data is definitely being received by odas_web.

Funchhy, 2020-11-24:
For me, if i specify IP and Port(in this case 5027), as well as configure it to be TCP + json in the config file i can get Data via:
```
import socket
import sys
import time

socket = socket.socket(socket.AF_INET, socket.SOCK_STREAM)
socket.bind(('', 5027))
socket.listen(5)
client, address = socket.accept()
print("{} connected".format( address ))
while True:
    response = client.recv(512)   
    print(response)
    time.sleep(1)
	
print ("close")
client.close()
stock.close()
```

Femtofirst, 2023-07-14:
> Hi,
>
> I wrote a little python script with a socket server to receive the SST data streamed by odaslive via a socket client. During some tests I noticed that with a buffer of 2048 bytes (actually down [...]

Hi! Four years later, I am trying to achieve the same function :) but kept on failing :(   
Could you share the python file, if you can still find it? I would really really appreciate that!