introlab/odas #183: Need help with saving audio input from individual microphones on Matrix

yihanhu97, 2019-11-01:
Hey guys, I've been using ODAS library for a research project for a while now and we would like to collect more data. 

Right now we are collecting the audio input generated through ODAS. e.g. 'separated.raw' & 'postfiltered.raw'. I understand that ODAS takes the individual audio input from 8 microphones on Matrix and then generates a combined audio file like just mentioned. For this research, we would also like to collect the raw audio input from the microphones. Thus by the end of each recording, we will have 10 audio files, 'separated.raw', 'postfiltered.raw', 'channel1.raw', ... ,'channel8.raw'

I wonder if this is possible and I'd appreciate it if anyone can give me any suggestions and point me in the right direction.

Thank you.

emil2099, 2020-01-05:
I wonder if this repo is active, as the project is quite amazing.

I also need to save individual input channels for a project. Is this possible?

Thanks!

FrancoisGrondin, 2020-01-05:
Hi there,

The project is still active :) I'm now starting a new position as a university professor, which will provide me with more time to work on ODAS and get students involved. The current version was released as beta, and we've received a lot of comments and suggestions. We could also eventually include some deep learning algorithms to make the system more up to date with the latest methods.

FrancoisGrondin, 2020-01-16:
I think this is not possible in this version (that's a shame...) but you could probably access the raw data loaded by the corresponding source module and just "dump" it in a file. If you look at this line:

https://github.com/introlab/odas/blob/master/src/source/src_hops.c#L363

You could dump the buffer in a file. That's a hack... I'm coding the new version and it would be much easier in the new version to obtain the raw and save it/send it somewhere.

Cheers,

yihanhu97, 2020-04-23:
Hi Francois,

Do you know if ODAS has a detecting range? I noticed that when I have a
source placed about 6 feet away, the sound separation and DOA estimation do
not work quite well.

Best,
Yihan