introlab/odas #187: I get only the activity of energy in a particular direction but I want to listen to the audio in that direction.

ajaygunalan, 2020-01-14:
You have mentioned that it is possible to record the sound in a fixed direction [here](https://github.com/introlab/odas/issues/158) and also that it is possible to change the directions via sockets dynamically [here](https://github.com/introlab/odas/issues/14). (not yet implemented though).

The issue is:
1. when I try to record the sound in a fixed direction. **I get only the activity of energy in that particular direction but I want to listen to the audio in that direction. How to do it?**
2. To change the values of x, y, z during run-time. Can i just add socket code at `odas/blob/master/demo/odaslive/parameters.c` and update the variable from a server. **Is it a good idea?** or could you explain how to use the `injector module` to update the directions?

FrancoisGrondin, 2020-01-16:
Hi,

1. To get the audio from that direction, you need to use the sound separation module.
2. You should not change the values of the parameters live, because once the system is initialized, it does not access these values anymore (mostly for dynamic memory allocation purpose). However, if you look at the "target" message, you can update the target over time to listen in a specific direction. There are many threads that address this issue. Let me know if you don't find what you are looking for.

Cheers,

ajaygunalan, 2020-01-20:
Hi Francois,

Thank you for the reply. 

I used the sss module. But, in static mode, I couldn't hear any audio. Only in dynamic mode, I was able listen.

This is the [config file](https://gist.github.com/ajaygunalan/c4ea692d70f8f08f32a63adb0291eaef). As said by you, I only change `add` to `dynamic` and comment out `myTarget` when used in dynamic mode.

You can find other files files such as the `separated.raw` for static and dynamic are [here](https://drive.google.com/open?id=14052cj_3DYRuy4hmS50uC0LxzQJzKjIM) along with a picture of my hardware setup.

ariga98, 2020-02-12:
Hello,

I was wondering if you had any progress in tracking audio sources in real time. If you have found any other threads that address this issue, or any found any other tips regarding the application that would be greatly appereciated!

Thanks

ajaygunalan, 2020-02-13:
Hey,

The Microphone Array Configuration was wrong in my config file. When I changed them,It works. But, only in Dynamic mode. Not able to listen to specific direction. (Static Mode). Not sure, what is the issue.

anrzej96, 2020-03-23:
I feel like I have a similar issue and i still can't find the solution.
Can you help me somehow?
https://github.com/introlab/odas/issues/194