introlab/odas #188: odas quits after 1 timestamp

dianalow, 2020-01-15:
Hi,
I'm new to ODAS, and using a UMA-8 microphone in Ubuntu. I got odas to execute without errors, but now it seems to quit after 1 timestamp.
```
arecord -l
**** List of CAPTURE Hardware Devices ****
card 0: Intel [HDA Intel], device 0: STAC9221 A1 Analog [STAC9221 A1 Analog]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 0: Intel [HDA Intel], device 1: STAC9221 A1 Digital [STAC9221 A1 Digital]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 0: Intel [HDA Intel], device 2: STAC9221 A1 Alt Analog [STAC9221 A1 Alt Analog]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 1: SpkUAC20 [miniDSP VocalFusion Spk (UAC2.0], device 0: USB Audio [USB Audio]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
```

the only change i made to the minidsp.cfg is the card number.
```
    # Input with raw signal from microphones
    interface: {
        type = "soundcard";
        card = 1;
        device = 0;
    }
```

```
bin/odaslive -c config/odaslive/minidsp_v2.cfg -v -s
+--------------------------------------------+
|    ODAS (Open embeddeD Audition System)    |
+--------------------------------------------+
| Author:  Francois Grondin                  |
| Email:   francois.grondin2@usherbrooke.ca  |
| Website: introlab.3it.usherbrooke.ca       |
| Version: 1.0                               |
+--------------------------------------------+
| + Initializing configurations...... [Done] |
| + Initializing objects............. [Done] |
| + Processing....................... {
    "timeStamp": 1,
    "src": [
        { "id": 0, "tag": "", "x": 0.000, "y": 0.000, "z": 0.000, "activity": 0.000 },
        { "id": 0, "tag": "", "x": 0.000, "y": 0.000, "z": 0.000, "activity": 0.000 },
        { "id": 0, "tag": "", "x": 0.000, "y": 0.000, "z": 0.000, "activity": 0.000 },
        { "id": 0, "tag": "", "x": 0.000, "y": 0.000, "z": 0.000, "activity": 0.000 }
    ]
}
[Done] |
| + Free memory...................... [Done] |
+--------------------------------------------+
|              Profiler summary              |
+--------------------------------------------+
| + Raw                                      |
|    - Source.......... 000.000 (001.20%)    |
|    - Connector....... 000.000 (000.15%)    |
| + Mapping                                  |
|    - Module.......... 000.000 (000.08%)    |
|    - Connector....... 000.000 (000.07%)    |
| + Resample                                 |
|    - Module.......... 000.000 (002.34%)    |
|    - Connector....... 000.000 (000.08%)    |
| + STFT                                     |
|    - Module.......... 000.000 (000.31%)    |
|    - Connector....... 000.000 (000.14%)    |
| + Noise                                    |
|    - Module.......... 000.000 (000.37%)    |
|    - Connector....... 000.000 (000.09%)    |
| + SSL                                      |
|    - Module.......... 000.001 (008.00%)    |
|    - Connector....... 000.000 (000.08%)    |
|    - Sink............ 000.000 (000.06%)    |
| + Target                                   |
|    - Injector........ 000.000 (000.04%)    |
|    - Connector....... 000.000 (000.05%)    |
| + SST                                      |
|    - Module.......... 000.000 (000.17%)    |
|    - Connector....... 000.000 (000.08%)    |
|    - Sink............ 000.000 (000.92%)    |
| + SSS                                      |
|    - Module.......... 000.000 (000.84%)    |
|    - Connector....... 000.000 (000.00%)    |
| + ISTFT                                    |
|    - Module.......... 000.000 (000.35%)    |
|    - Connector....... 000.000 (000.09%)    |
| + Resample                                 |
|    - Module.......... 000.000 (000.36%)    |
|    - Connector....... 000.000 (000.08%)    |
| + Volume                                   |
|    - Module.......... 000.000 (000.07%)    |
|    - Connector....... 000.000 (000.08%)    |
|    - Sink............ 000.000 (000.66%)    |
| + Classify                                 |
|    - Module.......... 000.000 (000.04%)    |
|    - Connector....... 000.000 (000.05%)    |
|    - Sink............ 000.000 (000.06%)    |
+--------------------------------------------+
| + TOTAL.............. 000.002 (016.95%)    |
|    - Module.......... 000.001 (012.92%)    |
|    - Connector....... 000.000 (001.14%)    |
|    - Source.......... 000.000 (001.20%)    |
|    - Sink............ 000.000 (001.69%)    |
+--------------------------------------------+

```

FrancoisGrondin, 2020-01-16:
You mean the behavior of ODAS changed over time with the same configuration file unchanged?

dianalow, 2020-05-21:
sorry for the slow reply. soundcard wasn't reading in directly and this is solved.

closed 2020-05-21 by dianalow

VioletIsSoCute, 2021-11-16:
I've bumped into similar problem. May I asked how you solved the soundcard problems?

Femtofirst, 2023-06-23:
Hi, I am testing with the same hardware device - I am wondering did you set up the UMA-8 before running ODAS? My terminal says: 'Source hops: Cannot set sample format: Invalid argument'  when I try to run the software, and I did not change anything in the .cfg file

VioletIsSoCute, 2023-06-23:
I've been off the project for a long time.
I do not record anything of UMA-8.