introlab/odas #189: Using "Static" in sst causes Segmentation fault.

hasagar97, 2020-01-16:
Hi, 
I am using odas with ReSpeaker 4-microphone array for raspberry pi, when I use SST with "dynamic" it works fine, but when I use it with "static" it crashes due to segmentation fault. 
PFA my config file

[respeaker4mic_arr.cfg.txt](https://github.com/introlab/odas/files/4070622/respeaker4mic_arr.cfg.txt)

hasagar97, 2020-01-16:
Okay, after changing the target to 
```cfg
target: (
        { 
           tag = "myTrackedSource"; x = 0.0; y = 0.0 z = 1.0 
        }
);
The segmentation fault issue is doesn't occur but I wan't to record sounds from a particular direction only.
Any ideas why it is happening or how to correct it @FrancoisGrondin ??
```

FrancoisGrondin, 2020-01-16:
Hey there,

I think you forgot a semicolon before the z variable.

Lemme know if it solves the issue

Cheers

hasagar97, 2020-01-20:
Hi,
I have updated the file and the problem still persists. Until and unless I keep z>=0.15 (I found this by trial and error) the program(odaslive) crashes due to segmentation fault. 

But I can work with this much amount of error(0.15m is not much), 
Thanks for the help.