introlab/odas #197: ODAS with Kinect DK

lingfei93, 2020-03-16:
Hi, can i check if ODAS will work on the Kinect DK? Been playing around with the cfg file, but I've been stuck at getting data to show up on the odas web server. Would appreciate a poke in the right direction!

xiaotaw, 2020-04-14:
I have test odas with Kinect DK, and it works fine. Here is my cofg file: https://github.com/xiaotaw/odas/blob/master/config/odaslive/azure_kinect_dk.cfg

However, I am not going to use odas web server. Instead, an ASR program will consume the postfiltered data via unix domain socket (act as IPC). 

Hope it helps.

keenanfitzharris, 2020-12-18:
I am trying to use my Kinect DK as well. I am using your config file which I edited the SST section so that the it does not utilize the Unix Domain Socket (I uncommented lines 253-256 and commented lines (257-261). 

However when I try to run odaslive I get the error 
"Source hops: Cannot set sample format: Invalid Argument:
Any idea as to what is wrong? 
Thanks