introlab/odas #202: Can I config the  ODAS  on the 8SoundsUsb?

jiguansghen, 2020-05-18:
[image]

I config ./odaslive -c matrix_voice.cfg -v in the odas_bin.Open another terminal and run npm start.The resullt as shown in the picture( + Initializing configurations...... [Done] |
| + Initializing objects............. [Done] |
| + Launch threads................... [Done] |
| + Threads running.................. Source hops: Cannot set sample rate: Invalid argument
).
I change the ip address and nbits,but it has no effect.

[matrix_voice .txt](https://github.com/introlab/odas/files/4643736/matrix_voice.txt)

ELELUABDULSALAM, 2022-01-31:
are you able to solve this

Chen-jia621, 2022-02-28:
I have the same question, do you solve it ?