introlab/odas #203: Tracking Signal from 6 mic array

Eric-cyber-1024, 2020-06-17:
Hi there, I am working a project on a service robot which could track a speaker from mic array. Right now, I am testing with the 6 mic circular Array kit for raspberry -Pi using the Odaslive and Odas studio. Observing the response from the terminal, I understand the return of the x,y,z and the activity. But is it possible to track the signal and set its ID for a permanent ID? 

For example, when I launch the ./odaslive , the terminal shows all zeros value. Once speaker starts speaking, it would register as the "first" speaker and it will keep tracking its coordinate and search for the speaker when it moves, specially if the speaker moves out of the mic array beam ?  And gives out the possible coordinate information for the robot to steer? 

Sorry for my poor English.

Thanks 
Eric

Eric-cyber-1024, 2020-06-18:
Another question .... Is it possible to generate another audio output for a instant playback, which it could be streaming into another interface and storing in a buffer? instead of using the file output?

closed 2020-07-10 by Eric-cyber-1024

Gaurav37, 2022-06-08:
Hi Eric, 
I am also working on a similar project. I want to get the distance of the sound source however it seems that SST does not work. It gives zero tracked sources.
How did you manage to solve this problem?

Thanks
Gaurav