introlab/odas #218: What "ms" and "ss" refers to in Sound Source Seperation module?

anrzej96, 2021-01-06:
Hello,

I have question - exactly same like the topic.
I've been doing my thesis using ODAS and i would like to describe methods used in SSS.
As i found, dss refers to Delay-and-Sum algorithm and DGSS - Gemoetry source seperation.
But i wasn't able to find anything about ss, and ms, which is really bad, cause 'ms' actually was the only mode that provided correct recordings in my project.

Can you help me out?
I found this https://arxiv.org/pdf/1812.00115.pdf - about SSL and SST it was very usful.
Also if it is possible to describe a little bit some parameters frome this modes.
The only thing that is written in configuration is - You should leave this parameter as x.y value which is not helpful at all.

Best Reagrds,
Jakub

anrzej96, 2021-01-06:
@GodCed @FrancoisGrondin - I'll really appreciate your help.

FrancoisGrondin, 2021-01-06:
Hi Jakub,

Here a paper you can refer to where we explain the ms mode: https://www.researchgate.net/publication/257523130_The_ManyEars_open_framework#fullTextFileContent

Please let me know if this is what you need!

Best,

anrzej96, 2021-01-06:
Thank you so much!

I'm sorry for bothering you via e-mail as well, I just didn't expect so quick answer.
I'll study it and let you know, but it looks like exactly what I need.

What about ss - is it also included in this paper?