introlab/odas #220: ReSpeaker Mic Array v2.0 (4 mic array) DOA is not matching with calculated azimuth from SSL json output

yugandharonchip, 2021-01-23:
Hi,

I used ReSpeaker Mic Array v2.0 (4 mic array) with ODAS and captured SSL json output in 4 directions(DOA = 0, 90, 180, 270) as in below picture. I used odas configuration file which is availble in odas/config/odaslive/respeaker_4_mic_array.cfg
[image]

I used below formula which is available in **odas_web\resources\js\graph.js** to calculate azimuth of each source from x, y coordinates in SSL json output.
`var az = Math.atan2(y,x);`

But, any one of the angles is not matching with ReSpeaker mentioned DOA. And also, in any direction when I am speaking, SSL output showing high energy(parameter E) in first source only. 

The calculated azimuths of all sources are as below.

**ReSpeaker  DOA = 0**
[143, 162, 146, -153]
[143, 146, 116, 112]
[143, 146, 116, 112]
[142, 146, -172, -167]
[146, -126, 0, 0]
[142, 162, 0, 0]
[146, 150, 0, 0]
[143, 146, 116, 112]
[142, 162, -172, -167]
[142, 162, -172, -167]

**ReSpeaker  DOA = 90**
[-142, -126, 0, 0]
[-142, -115, -54, -93]
[-142, -54, -54, -93]
[-142, -54, -54, -93]
[-142, -115, -54, -93]
[-142, -115, -54, -93]
[-142, -115, -54, -93]
[-142, -54, -54, -93]
[-142, -54, -54, -93]
[-142, -115, -54, -93]

**ReSpeaker  DOA = 180**
[-54, -45, -54, -93]
[-50, -47, -81, -32]
[-50, -47, -81, -32]
[-50, -47, -81, -32]
[-49, -49, -81, -32]
[-50, -49, -81, -32]
[-50, -49, -81, -32]
[-42, -54, -93, -87]
[-50, -49, -81, -32]
[-50, -49, -81, -32]

**ReSpeaker  DOA = 270**
[46, 27, 0, 0]
[46, 27, 18, 2]
[46, 27, 18, 2]
[46, 27, 18, 2]
[46, 27, 18, 2]
[46, 27, 18, 2]
[46, 27, 18, 2]
[46, 27, 18, 2]
[46, 27, 18, 2]
[46, 27, 18, 2]

How to understand the above azimuths? is it with respect to microphone position? How to compute ReSpeaker's DOA from above azimuths? If anybody have any ideas please suggest me.

Thanks,
Yugandhar

Qinyang-bit, 2021-11-12:
Hi,Yugandhar!

I have a similar issue. Did you solve the issue, and how? 

It's great if you can reply.
Thanks in advance,
Dongdong

StuartIanNaylor, 2022-02-22:
Should it not be respeaker_usb_4_mic_array.cfg

ELELUABDULSALAM, 2022-03-11:
Have you been able to solve this issue @yugandharonchip @Qinyang-bit