introlab/odas #222: How to get direction of audio through ODAS?

hltkts123, 2021-03-03:
Hi everyone!
I'm trying to do project that control a mobile robot by voice. I used Respeaker Core v2.0 to captured audio. I done with ODAS as the below figure.
Now I want to use the azimuth angle to control a mobile robot. I already have a complete robot.
So how do that?
[image]

P/s: I have tried retrieving messages via TCP / IP, maybe I coded wrong so I didnt receive anything.
I have tried edit file config to show information to terminal and I got some parameters as x, y, z, energy...etc.
[image]

doumdi, 2021-03-09:
https://github.com/introlab/odas_ros/tree/redesign_modular_v2/sound_tracking

peisong0109, 2021-03-12:
> Hi everyone!
> I'm trying to do project that control a mobile robot by voice. I used Respeaker Core v2.0 to captured audio. I done with ODAS as the below figure.
> Now I want to use the azimuth angle to [...]

How did you get these parameter information, includes: x, y, z, energy?
Probably your method is really helpful for me.

hltkts123, 2021-03-15:
> https://github.com/introlab/odas_ros/tree/redesign_modular_v2/sound_tracking

It's amazing! I have cloned it to my laptop and following readme.md. However when I "catkin_make", it showed error as below:
/home/freedom/Pictures/Screenshot from 2021-03-15 09-13-46.png
I tried to find it, but it failed

hltkts123, 2021-03-15:
> How did you get these parameter information, includes: x, y, z, energy?
> Probably your method is really helpful for me.

Currently I don't have a way to get the parameters. I'm trying use methor suggested by doumdi above

ganyihangxingyihang, 2021-07-22:
How to start odas_ros,after installation is complete,i dont know how to do

Qinyang-bit, 2021-11-12:
Hi, I am so so excited that you have get  these parameter information, these are just what I need.
Can you tell me how you got them? That will help me a lot!

It's great if you can reply,
Thanks in advance!
Dongdong

Qinyang-bit, 2021-11-14:
> How did you get these parameter information, includes: x, y, z, energy? Probably your method is really helpful for me.

May I see the full contents of your terminal? That will be so helpful for me! 
Thank you so much！！