introlab/odas #226: Socket error: "Sink pots: Cannot connect to server"

LudovicDeMatteis, 2021-04-05:
Hello,
I've been working with odas_ros and the 16SoundsUSB card. It is plugged in my computer and I ran the odas ros package with the following configuration file:
`version = "2.1";

# Raw

raw: 
{

    fS = 16000;
    hopSize = 512;
    nBits = 32;
    nChannels = 16; 

    # Input with raw signal from microphones
    interface: {
        type = "soundcard";
        card = 1;
        device = 0;
    }

}

# Mapping

mapping:
{
    map: (1, 2, 3, 4, 5, 6, 7, 8);
}

# General

general:
{

    epsilon = 1E-20;

    size:   #for fft calculation
    {
        hopSize = 128;      #shift size of the cross fft
        frameSize = 256;    #size of each fft of the cross fft
    };

    samplerate:
    {
        mu = 16000;
        sigma2 = 0.01;
    };

    speedofsound:
    {
        mu = 343.0;
        sigma2 = 25.0;
    };

    mics = (

        # Microphone 1 #TODO
        {
            mu = ( 0, -0.09, -0.05 );
            sigma2 = ( +1E-6, 0.0, 0.0, 0.0, +1E-6, 0.0, 0.0, 0.0, +1E-6 );
            direction = ( 0, 1, 0 );
            angle = ( 80.0, 100.0 );
        },

        # Microphone 2 #TODO
        {
            mu = ( 0.034, 0.083, 0.05 );
            sigma2 = ( +1E-6, 0.0, 0.0, 0.0, +1E-6, 0.0, 0.0, 0.0, +1E-6 );
            direction = ( 0.3826834324, 0.9238795325, 0 );
            angle = ( 80.0, 100.0 );
        },

        # Microphone 3 #TODO
        {
            mu = ( 0.064, 0.064, -0.05 );
            sigma2 = ( +1E-6, 0.0, 0.0, 0.0, +1E-6, 0.0, 0.0, 0.0, +1E-6 );
            direction = ( 0.7071067812, 0.7071067812, 0 );
            angle = ( 80.0, 100.0 );
        },

        # Microphone 4 #TODO
        {
            mu = ( 0.083, 0.034, 0.05 );
            sigma2 = ( +1E-6, 0.0, 0.0, 0.0, +1E-6, 0.0, 0.0, 0.0, +1E-6 );
            direction = ( 0.9238795325, 0.3826834324, 0 );
            angle = ( 80.0, 100.0 );
        },

        # Microphone 5 #TODO
        {
            mu = ( 0.083, -0.034, -0.05 );
            sigma2 = ( +1E-6, 0.0, 0.0, 0.0, +1E-6, 0.0, 0.0, 0.0, +1E-6 );
            direction = ( 0.9238795325, -0.3826834324, 0 );
            angle = ( 80.0, 100.0 );
        },

        # Microphone 6 #TODO
        {
            mu = ( 0.064, -0.064, 0.05 );
            sigma2 = ( +1E-6, 0.0, 0.0, 0.0, +1E-6, 0.0, 0.0, 0.0, +1E-6 );
            direction = ( 0.7071067812, -0.7071067812, 0 );
            angle = ( 80.0, 100.0 );
        },

        # Microphone 7 #TODO
        {
            mu = ( 0.034, -0.083, -0.05 );
            sigma2 = ( +1E-6, 0.0, 0.0, 0.0, +1E-6, 0.0, 0.0, 0.0, +1E-6 );
            direction = ( 0.3826834324, -0.9238795325, 0 );
            angle = ( 80.0, 100.0 );
        },

        # Microphone 8 #TODO
        {
            mu = ( 0, -0.09, 0.05 );
            sigma2 = ( +1E-6, 0.0, 0.0, 0.0, +1E-6, 0.0, 0.0, 0.0, +1E-6 );
            direction = ( 0, -1, 0 );
            angle = ( 80.0, 100.0 );
        }

    );

    # Spatial filters to include only a range of direction if required
    # (may be useful to remove false detections from the floor, or
    # limit the space search to a restricted region)
    spatialfilters = (
        {
            direction = ( +0.000, +0.000, +1.000 );
            angle = (80.0, 90.0);
        }
    );

    nThetas = 181;
    gainMin = 0.25;

};

# Stationnary noise estimation

sne:
{
    
    b = 3;
    alphaS = 0.1;
    L = 150;
    delta = 3.0;
    alphaD = 0.1;

}

# Sound Source Localization

ssl:
{

    nPots = 2;
    nMatches = 10;
    probMin = 0.5;
    nRefinedLevels = 1;
    interpRate = 4;

    # Number of scans: level is the resolution of the sphere
    # and delta is the size of the maximum sliding window
    # (delta = -1 means the size is automatically computed)
    scans = (
        { level = 2; delta = -1; },
        { level = 4; delta = -1; }
    );

    # Output to export potential sources
    potential: {

        format = "json";

        interface: {
            type = "socket";
            ip = "127.0.0.1";
            port = 9001;
        };

        #format = "undefined";

        #interface: {
        #   type = "blackhole";
        #};

    };

};

# Sound Source Tracking

sst:
{

    # Mode is either "kalman" or "particle"

    mode = "kalman";

    # Add is either "static" or "dynamic"

    add = "dynamic";

    # Parameters used by both the Kalman and particle filter

    active = (
        { weight = 1.0; mu = 0.3; sigma2 = 0.0025 }
    );

    inactive = (
        { weight = 1.0; mu = 0.15; sigma2 = 0.0025 }
    );

    sigmaR2_prob = 0.0025;
    sigmaR2_active = 0.0225;
    sigmaR2_target = 0.0025;
    Pfalse = 0.1;
    Pnew = 0.1;
    Ptrack = 0.8;

    theta_new = 0.9;
    N_prob = 5;
    theta_prob = 0.8;
    N_inactive = ( 150 );
    theta_inactive = 0.9;

    # Parameters used by the Kalman filter only

    kalman: {

        sigmaQ = 0.001; #bigger=more reactive / smaller=more robust the noise
        
    };
   
    # Parameters used by the particle filter only

    particle: {

        nParticles = 1000;
        st_alpha = 2.0;
        st_beta = 0.04;
        st_ratio = 0.5;
        ve_alpha = 0.05;
        ve_beta = 0.2;
        ve_ratio = 0.3;
        ac_alpha = 0.5;
        ac_beta = 0.2;
        ac_ratio = 0.2;
        Nmin = 0.7;

    };

    target: ();

    # Output to export tracked sources
    tracked: {

        #format = "json";

        #interface: {
        #    type = "socket";
        #    ip = "127.0.0.1";
        #    port = 9000;
        #};

	format = "undefined";

	interface: {
            type = "blackhole";
        };

    };

}

# Sound Source Separation

sss:
{
    
    # Mode is either "dds", "dgss" or "dmvdr"

    mode_sep = "dds"; #delay and sum
    mode_pf = "ms";

    gain_sep = 2.0;
    gain_pf = 10.0;

    dds: {

    };

    dgss: {

        mu = 0.01;
        lambda = 0.5;

    };

    dmvdr: {

    };

    ms: {

        alphaPmin = 0.07;
        eta = 0.5;
        alphaZ = 0.8;
        thetaWin = 0.3;
        alphaWin = 0.3;
        maxAbsenceProb = 0.9;
        Gmin = 0.01;
        winSizeLocal = 3;
        winSizeGlobal = 23;
        winSizeFrame = 256;

    };

    ss: {

        Gmin = 0.01;
        Gmid = 0.5;
        Gslope = 10.0;

    };

    separated: { #packaging and destination of the separated files

        fS = 44100;
        hopSize = 256;
        nBits = 16;

        #interface: {
        #    type = "socket";
        #    ip = "127.0.0.1";
        #    port = 9001;
        #}  

	interface: {
            type = "blackhole";
        }     
        

    };

    postfiltered: { #packaging and destination of the post filtered files

        fS = 44100;
        hopSize = 256;
        nBits = 16;

        interface: {
            type = "blackhole";
        }

        

    };

}

classify:
{

    frameSize = 1024;
    winSize = 3;
    tauMin = 32;
    tauMax = 200;
    deltaTauMax = 7;
    alpha = 0.3;
    gamma = 0.01;
    phiMin = 0.5;
    r0 = 0.2;

    category: {
        format = "undefined";

        interface: {
            type = "blackhole";
        }
    }
}`

The part of the configuration that is most interesting for this issue is:
`    potential: {

        format = "json";

        interface: {
            type = "socket";
            ip = "127.0.0.1";
            port = 9001;
        };

        #format = "undefined";

        #interface: {
        #   type = "blackhole";
        #};

    };`
    
When running the node that launches Odas I get the following error: Sink pots: Cannot connect to server
That certainly means that the card cannot acces the socket at IP 127.0.0.1:9001. I'm not experienced with networks so I don't know how to fix this issue. Any help would be welcome.

Many thanks

Ludovic

doumdi, 2021-04-06:
@LudovicDeMatteis, Can you try to use the latest (just released last week) odas_ros code here: https://github.com/introlab/odas_ros ? Use the sample configuration in here to start : https://github.com/introlab/odas_ros/blob/main/config/configuration.cfg

doumdi, 2021-04-06:
@LudovicDeMatteis, could you describe in details what you are trying to achieve? Basically, the odas_core_node needs to connect to a python node and you seem to have a configuration mismatch  (wrong port?). We simplified the setup in the new version parsing directly only one configuration file avoiding this problem.

LudovicDeMatteis, 2021-04-06:
@doumdi I used the latest version of odas_ros and the configuration file you suggested. I seems to have gotten past the socket error. I now encounter a python error:
`OdasServerNode instance has no attribute '_sst_frame_id'
OdasServerNode instance has no attribute '_ssl_frame_id'
`
It most likely happens in the odas_server_node.py file in "send_sst" line 188:
`odas_sst_array_stamped_msg.header.frame_id = self._sst_frame_id` where sst_frame_id may not have been initialized.

To describe further what I want to achieve: I have a 16SoundUSB card connected to my computer and I'd like to get the informations about the sources positions in ROS. So I would use oda_ros to publish in topics that I can read from in other nodes.

LudovicDeMatteis, 2021-04-06:
I defined a frame_id that suits my aplication and I past the error. I now got a new one concerning pcl2.
`[ERROR] [1617719530.472705]: bad callback: <bound method OdasVisualizationNode._ssl_cb of <__main__.OdasVisualizationNode instance at 0x7f04ed2e0500>>
Traceback (most recent call last):
  File "/opt/ros/melodic/lib/python2.7/dist-packages/rospy/topics.py", line 750, in _invoke_callback
    cb(msg)
  File "/home/ludovic/gitLab/speech_dialogue_new/src/odas_ros/scripts/odas_visualization_node.py", line 82, in _ssl_cb
    pcl = pcl2.create_cloud(header,fields,cloud_points)
NameError: global name 'pcl2' is not defined
`

doumdi, 2021-04-06:
@LudovicDeMatteis You need to put the odas_ros package in the src directory of the catkin workspace and do a "catkin_make" to generate the adequate ROS messages. Then, you should launch (with roslaunch) your project by creating a launch file similar to this one : https://github.com/introlab/odas_ros/blob/main/launch/odas.launch.

oliroy92, 2021-04-06:
Hi @LudovicDeMatteis, 

I just pushed a new version of odas_ros with modifications to fix these errors. Can you try again after having pulled?

Thanks!

doumdi, 2021-04-06:
@LudovicDeMatteis Please tell us if you are successful with the patch. Thanks!

LudovicDeMatteis, 2021-04-07:
@doumdi @oliroy92  Everything is working fine now. I can read the topics and visualize on rviz which is exactly what I wanted. Thanks a lot for your help. Have a nice day

LudovicDeMatteis, 2021-04-12:
Hi, I'm reopening this issue because I facing new problems. 

To summarize, I got a Ubuntu 18.04 computer with a 16SoundUSB card plugged in (USB). Using the given config file I can run odas_ros and get published topics. However, looking at the topic `odas/sst/sources` I do not get any sources detected. 

I tried to set the soundcard as the acquisition device in Audacity to verify that it gets something and sound is indeed detected.

Problems may come from the configuration file. I ran `arecord -l` to verify that the card number is correct and I got the following output:
```
**** List of CAPTURE Hardware Devices ****
card 0: PCH [HDA Intel PCH], device 0: ALC255 Analog [ALC255 Analog]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 0: PCH [HDA Intel PCH], device 2: ALC255 Alt Analog [ALC255 Alt Analog]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 1: D20 [16SoundsUSB Audio 2.0], device 0: USB Audio [USB Audio]
  Subdevices: 0/1
  Subdevice #0: subdevice #0
```
Hence the configuration file interface being: 
```
raw: 
{
    fS = 16000;
    hopSize = 512;
    nBits = 32;
    nChannels = 16; 
    # Input with raw signal from microphones
    interface: {    #"arecord -l" OR "aplay --list-devices" to see the devices
        type = "soundcard_name";
        devicename = "hw:CARD=1,DEV=0";
    }
}
```

I don't know what else to test or check. 

Thanks,
Ludovic

oliroy92, 2021-04-12:
Hi @LudovicDeMatteis,

Normally, with the latest release of odas_ros, the topic for the sound source tracking (sst) should be published on the topic `/odas/sst`. If you are able to launch odas_ros and it doesn't crash, it should mean that your sound card is recognized. When you do launch odas_ros with `roslaunch`, you should be able to set the `rviz` parameter to true to see the published sst in Rviz. You can also specify a configuration file path if you want to. 

``` 
roslaunch odas_ros odas.launch rviz:=true configuration_path:=~/catkin_ws/path_to_config_file/.../configuration.cfg
```

If you execute this, you should be able to see the published topic for the sound source tracking and sound source localization (if enabled in configuration file) visually:

[image]

If you can't see anything, it may be because of a incorrect configuration file. Your sound card configuration looks to be correct, so it could be the socket configuration.

Can you try this and see if data appears on rviz? If you have a doubt about your configuration file, you can use the one that is on the repository, it should work (only change your sound card number and device number like you did).

Olivier

LudovicDeMatteis, 2021-04-12:
Hi @oliroy92 

I ran the command to run rviz with odas.launch and I indeed get points from the ssl topic. However there is no arrow showing up even when there is a sound source. When the sound source gets louder the number of points reduces so I assume the card is indeed getting some sound and there is less potential sources when the main source is clearer.

 In the terminal I get some messages: `Extra data: line 8 column 1 - line 15 column 1 (char 168 - 336)`
 
[image]

This may be related to the fact that Sound Source Tracking is not giving results. 

Note that I changed the configuration file to have the correct positions and orientations on my 8 microphones and that I only map microphones 1 to 8:

> mapping:
{
    map: (1, 2, 3, 4, 5, 6, 7, 8);
}

Thanks,
Ludovic

LudovicDeMatteis, 2021-04-13:
The dots of the ssl point cloud are mostly red to yellow in the direction of the source unlike in your picture where they are mostly green. There may be a problem of intensity?

The topic `/odas/sst` is published but the "sources" field is empty :
[image]

Here is a picture of the microphone array configuration. Unfortunately I don't have a CAD file available. The microphone numbers on the pictures are not corresponding to how they really are plugged in the card but I double-checked the microphones positions (the mu) and their directions in the config file to verify that it is correct (with microphone i being plugged at port i on the card).

Attached is the config file I used.

[configuration.zip](https://github.com/introlab/odas/files/6302465/configuration.zip)

doumdi, 2021-04-13:
@LudovicDeMatteis, can you try increasing the gains on each microphone (there is a blue multi-turn potentiometer for that)? To help, I use Audacity to make sure I turn in the right direction to increment the gain & amplitude while making an acquisition. Please try the maximum gain for the 8 microphones you use. You should hear a "click" from the potentiometer when you reach the end of the range.

oliroy92, 2021-04-13:
@LudovicDeMatteis,

In addition to what @doumdi said, here are a couple more things to look at:

> Here is a picture of the microphone array configuration. 

Could you make sure to include that picture in your comment?

From your configuration file, it looks like the microphone configuration is a semi-circle with microphones of alternating heights. Is that correct? Also, you seem to have configured your file with 16 microphones but have only 8 mapped. Is there a reason why you would not use all 16? And if there is only 8 microphones on your array, maybe you could remove microphones 9 to 16 from your configuration file.

If I'm not mistaken, there seems to be an error for the direction of Microphone #5 in your configuration file. 
[image]
The direction should be set to `direction = ( 0.9238795325, 0.3826834324, 0 );`

Olivier

LudovicDeMatteis, 2021-04-14:
@doumdi @oliroy92 the microphone gains are already set to the maximum.

> Could you make sure to include that picture in your comment?

Sorry, I didn't see that the picture was not attached. Here it is now: 
[image]

Indeed it is a semi circle with microphones of alternating heights. I had a couple mistakes in my configuration file that I fixed now. 
We use only 8 microphone because of constraints on the robot it will be implemented on. The laboratory will certainly use a 8SoundsUSB instead of the 16SoundsUSB in the future.

Thus only 8 microphones are mapped. When I try to remove microphones 10 to 16 from the configuration file it seems to work better (points are plotted with better intensity in the points cloud) but I cannot remove microphone 9 from the configuration file otherwise it raises the error `Sink tracks: Cannot connect to server` and `Sink hops: Cannot connect to server`. These errors disappear as soon as I add a 9th microphone to the configuration file. This seems to happen even if I map microphones 1-7 instead of 1-8. 

When I remove microphones from the config file I also got more often the message I mentioned previously: 

> In the terminal I get some messages: `Extra data: line 8 column 1 - line 15 column 1 (char 168 - 336)`

However, the red arrow is now appearing and following the source. The problem was certainly a intensity problem that have been solved by removing some microphones from the configuration file.

LudovicDeMatteis, 2021-04-21:
@doumdi @oliroy92 Hi, I got one more question. Is odas_ros capable of separating two sources and gives both their positions? I suppose that the SSL topic can be used to get the most probable positions of two (or more) source but will the SSS et SST topics give distinct positions and audio?
Thanks,
Ludovic

doumdi, 2021-04-21:
@LudovicDeMatteis , yes ODAS can track and separate up to 4 sources.

mamadouDembele, 2021-04-29:
Hi I am currently working with the respeaker 4 mic array voice interface. I'm trying to find the location of the sound source thanks to the odas_ros packages. After compiling this package, when I run the launch, I first get this error '' Sink tracks: Cannot connect to server''
[image]

I said to myself as I had not launched odas studio, the error is surely due to that. So I first launched odas studio (npm start) then the launch file (roslaunch odas_ros odas.launch) I get a new error "error: [Errno 98] Address already in use".
[image]

Could someone help me

closed 2021-06-07 by LudovicDeMatteis