introlab/odas #229: odaslive segmentation faults after running for some time

somewacko, 2021-04-17:
I am running `odaslive` on a Raspberry Pi 4 with a ReSpeaker 4 Mic Array, but after some time (anywhere from a few seconds to several minutes later) it will crash with a segmentation fault. I am using the provided config at "config/odaslive/respeaker_4_mic_array.cfg" and have only changed the IP address.

I've so far ruled out the driver since I can record audio without crashing, and this seems to happen both when connecting over LAN or localhost to a background process.

Any guidance for how I could debug this?

(awesome project btw!)

suegy, 2021-06-12:
I have the same issue running it on a Pi4 using the 6 mic array. 
I tried the web server and writing to file so it cannot be the connection.

MarkSttc, 2022-04-19:
Same issue happened to me :C

DmitryP11, 2022-04-19:
have you disabled this?
interface: {
            type = "file";
            path = "postfiltered.raw";
        } 

        interface: {
            type = "file";
            path = "separated.raw";
        }

MarkSttc, 2022-04-19:
@DmitryP11 May I know how to disable this interface?.  Can we just change type = "undefined" and path = "blackhole"

MarkSttc, 2022-04-20:
@DmitryP11 It works fine now. Thank you so much for your help :D. 

However, Is there any chance u can explain to me what happened behind this segmentation fault?.

DmitryP11, 2022-04-20:
I think there is memory overflow in dumping raw data, but this is just a wild guess