introlab/odas #232: Hardware recommendations

prospero12, 2021-05-18:
Very interested in using ODAS for sound source localisation and tracking in, and across rooms with a fair amount of echo and ambient noise; thus would much appreciate any recommendations on hardware/mic arrays to use.  Some options that people seem to have used, are

1. MATRIX creator
2. ReSpeaker v2 variants
3. 16SoundsUSB

Cost may also play a part, but pros and cons, and any recommendations (or reasons to avoid!) would be wonderful.  The key purpose (for my personal project) is to know where sound leakage and transfer is occurring.

nyxrobotics, 2021-06-14:
I bought XK-USB-MIC-UF216 and Respeaker V2 and UMA-8 and ps3-eye.
Between XK-USB-MIC-UF216 and Respeaker V2, XK-USB-MIC-UF216 had better beam forming performance.
PS3eye has good sound quality but very poor sound source localization performance.
UMA-8 will be tested in the future.

I read this article.
https://arxiv.org/pdf/2103.03954.pdf

nyxrobotics, 2021-06-18:
The UMA-8 was similar in performance to the XK-USB-MIC-UF216.
I ordered MATRIX creator a while ago. I'm looking forward to receiving it.

prospero12, 2021-06-21:
Many thanks @nyxrobotics for the recommendations.  I've since bought a ReSpeaker v2, and agree, it doesn't quite achieve what I had intended.  Will investigate the UMA-8 if able.

nyxrobotics, 2021-06-28:
The MATRIX creator seems to be for use with the Raspberry Pi and does not have a USB connection.
It doesn't seem to be suitable for my use.
The ReSpeaker v2 and the UMA-8 are quite different in performance, so you may want to try them out.

Femtofirst, 2023-06-23:
Hi guys, did you use Raspberry PI or a Linux system to run the UMA-8? Any difference between this 2 options? And did you set up the device before running ODAS?

nyxrobotics, 2023-06-23:
> Hi guys, did you use Raspberry PI or a Linux system to run the UMA-8? Any difference between this 2 options? And did you set up the device before running ODAS?

You have to write firmware with windows.
I referenced [this](https://www.minidsp.com/forum/usb-mic-array/13684-uma-8-mic-array-driver-downloads) article.
1. Create account in [minidsp](https://www.minidsp.com/userdownloads/usb-mic-array-series) website
2. Download a file name like `micArray_VF_v1.3_mac_dfu_v2_20.zip`
3. Install windows DFU driver with `Windows_Driver/miniDSP_UAC2_v4.82.0_2020-06-09_setup.exe`
4. Reboot windows
5. Run `miniDSP_UAC2_DFU_Win/miniDSPUAC2Dfu.exe`
6. `Brouse -> micArray_vf_raw_v1.3_up.bin`
7. Start

Femtofirst, 2023-06-23:
> You have to write firmware with windows. I referenced [this](https://www.minidsp.com/forum/usb-mic-array/13684-uma-8-mic-array-driver-downloads) article.
>
>     1. Create account in [...]

Thanks a lot! I am new to this so here is another question - after writing the firmware with windows, can I unplug it and use it on another Ubuntu computer? - Will the settings remain in the device and wont be affected by the change of the system?

nyxrobotics, 2023-06-23:
> Thanks a lot! I am new to this so here is another question - after writing the firmware with windows, can I unplug it and use it on another Ubuntu computer? - Will the settings remain in the device and wont be affected by the change of the system?

Yes. It is successful if the 8ch microphone device is recognized when connected to ubuntu.

Femtofirst, 2023-06-23:
> Yes. It is successful if the 8ch microphone device is recognized when connected to ubuntu.

Thank you so much! I am also using Ubuntu instead of raspberry PI - it's weird that my device is recognized when connected to the pc, but when I was trying to run ODAS with minidsp.cfg the system says 'Source hops: Cannot set sample format: Invalid argument'.

Do you maybe know what is the problem? And which configuration file did you use to run odas?

nyxrobotics, 2023-06-23:
My config here. Please copy and paste it.
<details>
  <summary>minidsp_uma8.cfg</summary>

  ```
# Configuration file for XMOS circular sound card

version = "2.1";

# Raw

raw:
{

    fS = 48000;
    hopSize = 512;
    nBits = 32;
    nChannels = 8;

    # Input with raw signal from microphones
    interface: {
        type = "soundcard";
        card = 2;
        device = 0;
    }

}

# Mapping

mapping:
{

    map: (1, 2, 3, 4, 5, 6, 7);

}

# General

general:
{

    epsilon = 1E-20;

    size:
    {
        hopSize = 128;
        frameSize = 256;
    };

    samplerate:
    {
        mu = 16000;
        sigma2 = 0.01;
    };

    speedofsound:
    {
        mu = 343.0;
        sigma2 = 25.0;
    };

    mics = (
   
        # Microphone 0
        {
            mu = ( +0.000, +0.000, +0.000 );
            sigma2 = ( +0.000, +0.000, +0.000, +0.000, +0.000, +0.000, +0.000, +0.000, +0.000 );
            direction = ( +0.000, +0.000, +1.000 );
            angle = ( 90.0, 100.0 );
[... 274 lines cut ...]
{

    frameSize = 4096;
    winSize = 3;
    tauMin = 88;
    tauMax = 551;
    deltaTauMax = 20;
    alpha = 0.3;
    gamma = 0.05;
    phiMin = 0.5;
    r0 = 0.2;

    category: {

        format = "undefined";

        interface: {
            type = "blackhole";
        }

    }

};
  ```
</details>

nyxrobotics, 2023-06-23:
The ```pactl list sources``` command can be used to check the number of channels, data format (16bit/32bit), and sampling rate of the sound card being used. This should be reflected in the configuration file.

Femtofirst, 2023-06-23:
Thank you so much! That is very helpful! I have followed your tutorial and now it reports something different: 

Sink pots: Cannot connect to server
Sink tracks: Cannot connect to server

Guess it is some kind of network error?

nyxrobotics, 2023-06-23:
> 

> Thank you so much! That is very helpful! I have followed your tutorial and now it reports something different:
> 
> Sink pots: Cannot connect to server Sink tracks: Cannot connect to server
> 
> Guess it is some kind of network error?

odaslive doesn't work with that error. Let me know if it works.
In the meantime, you can try odas_web.

nyxrobotics, 2023-06-23:
In the configuration file, you need to specify the port on which the sound card is recognized. Use the "arecord -l" command to see which port the device is recognized on and edit the number in "interface".

Femtofirst, 2023-06-27:
> In the configuration file, you need to specify the port on which the sound card is recognized. Use the `arecord -l` command to see which port the device is recognized on and edit the number in `interface`.

Thank you! Finally odas is running! Now I just need to get the GUI work