introlab/odas #233: Multiple source tracking not working

LudovicDeMatteis, 2021-06-07:
Hi,
I tried to test multiple source tracking on the latest ODAS version and I can't have it working. I used odas_web to have a visualization of the source tracking and even when there are two sources, odas seems to only track one. I attach below the figure to illustrate that.
[image]

Here is the config file that I am using: [configuration_circle.zip](https://github.com/introlab/odas/files/6607928/configuration_circle.zip)
I only got 8 microphones plugged in for now but I don't think that is an issue.

Thanks for the help, 

Ludovic

nyxrobotics, 2021-06-14:
I'm not an expert on odas, but what if you set nPots to 4?
```
# Sound Source Localization

ssl:
{
    nPots = 1;
    nMatches = 10;
    probMin = 0.5;
    nRefinedLevels = 1;
    interpRate = 4;

```

LudovicDeMatteis, 2021-06-21:
Hi @nyxrobotics, thank you for the suggestion. Unfortunately that does not solve the problem for me.

closed 2024-06-12 by LudovicDeMatteis