introlab/odas #235: run odaslive from command line

lembit1, 2021-06-14:
Hi,
I would like to use odaslive to process .raw input files without using the odas_web GUI. Is there a commandline form to do this?
Thanks!

subhalingamd, 2021-06-29:
Hi! I had a similar question few days back and I have found a solution recently (so thought I can share it here).

In the configuration file, modify the `interface` in `raw` section (as below) to get input from a `.raw` file.
```
raw: 
{
    ...

    interface: {
        type = "file";
        path = "mics.raw";      
    };

};
```

The outputs of the module can be obtained in *terminal* or *file* by changing the `interface` options under **each** module in the configuration file.
```

        # eg for terminal output
        interface: {
            type = "terminal";
        };

```

```

        # eg for file output
        interface: {
            type = "file";
            path = "tracked.json";
        };    

```

Hope it helps!

vfina, 2021-06-29:
Hi @subhalingamd!
Following your instructions is it possible to process .wav files? 
I've a certain number of audio tracks and I need to evaluate the DoA with ODAS. Do you have any hints?
I initially thought about setting the configuration file parameters like this:
```ini
raw: {
    fS = 16000;
    hopSize = 128;
    nBits = 16;
    # The audio files to be evaluated have 4 channels
    nChannels = 4;
    # Input with raw signal from file
    interface: {
        type = "file";
        path = "path_to_wav_file";
    }
}
```
Thanks!

lembit1, 2021-06-29:
hi @subhalingamd ! Thank you for pointing out how to edit the .config file to select data source and sink 👍 . The question is how to run  ODAS core without using the ODAS Web interface?

lembit1, 2021-06-29:
Hi @vfina . The .raw and .wav formats are very similar. The .wav format does have a metadata header in <INFO> chunk https://en.wikipedia.org/wiki/WAV. Everything after the .wav header is the .raw data. For manual preprocessing, you could use a hex editor to strip out the .wav header. http://soundfile.sapp.org/doc/WaveFormat/ is a user-friendly .wav format discussion

For a automated solution, I suggest 2 options:
1) write a separate script to remove the .wav header
2) dive in to the ODAS Library and write a .wav ingestion capability

Would love to either approach implemented!

subhalingamd, 2021-06-30:
> hi @subhalingamd ! Thank you for pointing out how to edit the .config file to select data source and sink 👍 . The question is how to run ODAS core without using the ODAS Web interface?

Hi @lembit1. You can use the following command. 
```
/path/to/odaslive -c /path/to/myConfigFile.cfg
```
The `odaslive` executive would normally be in `bin/` after compilation. 

---
PS: fyi the steps for setting up ODAS is mentioned in Wiki

subhalingamd, 2021-06-30:
> Hi @subhalingamd!
> Following your instructions is it possible to process .wav files?
> I've a certain number of audio tracks and I need to evaluate the DoA with ODAS. Do you have any hints?
> I initially thought about setting the configuration file parameters like this:
> 

Hi @vfina! I don't think `.wav` files are supported directly. @lembit1 has mentioned one approach. An easier way (imo) would be to use libraries that can be used on command-line. I have been using `sox` and the following commands:

```
# merge wav files of each channel (here, 3 channels) into a single multi-channel wav file (out.wav)
sox -M wav_mic1.wav wav_mic2.wav wav_mic3.wav  out.wav

# convert wav to raw (modify the values in the arguments accordingly)
sox out.wav --type raw --bits 16 --channels 3 --rate 48000 --encoding signed-integer out.raw
```