introlab/odas #238: Cannot set sample format: Invalid argument

InnovationCET, 2021-07-15:
after successfully installed ODAS on ubuntu 20.04, with freshly installed latest 6-ch firmware for respeaker microphone,
I failed to start odaslive. Launch it using the ODAS studio also not seems to work :/

odas git:(master) ✗ bin/odaslive -c config/odaslive/respeaker_usb_4_mic_array.cfg -v
+--------------------------------------------+ 
|    ODAS (Open embeddeD Audition System)    |
+--------------------------------------------+
| Author:  Francois Grondin                  |
| Email:   francois.grondin2@usherbrooke.ca  |
| Website: introlab.3it.usherbrooke.ca       |
| Version: 1.0                               |
+--------------------------------------------+
| + Initializing configurations...... [Done] |
| + Initializing objects............. [Done] |
| + Launch threads................... [Done] |
| + Threads running.................. Source hops: Cannot set sample format: Invalid argument

InnovationCET, 2021-08-22:
update:
I've managed to solve the issue, and of course - thanks a lot it is a great project! 
 it was in parsing of the configuration file. is it a conventional configuration file?

closed 2021-08-22 by InnovationCET

Chen-jia621, 2022-02-28:
Could you tell me how to solve this problem? thanks!

ELELUABDULSALAM, 2022-02-28:
Check your sound card type

deeprobo-dev, 2022-08-03:
sudo aplay -l run this in terminal to know what is your card no and put it in odas.cfg file