introlab/odas #241: Camera viewer

fanman2014, 2021-08-07:
Hi,

I wish to superimpose the sound locations onto a live webcam stream. In other words, I would like the peak sound energy locations to be located on a live video stream showing the sound source. What is the best way to do this?

BTW, I also noticed that the app has a "camera" menu item. What does this really do? When I select it nothing happens.

Look forward to your help.

thanks

Mark.

fanman2014, 2021-08-12:
Ok, I've written a Python script below to read in ODAS data based on Port 9001 / SSL. I've tried to invoke Odaslive via a command in Python and then connecting to the socket. I try to establish a connection using the terminal command "nc 10.0.0.17 9001". Sometimes I cannot connect to the server, other times if I get a connection, no data is coming through. I'm running this on a Raspberry Pi 4. Please help!

import json
import socket
import threading
import io
import sys
import time
import os
import subprocess

## Start odaslive process

cmd ='./odaslive -c respeaker_6_mic_array.cfg'
os.system(cmd)

## Start recording odas data via TCP/IP socket

port = 9001
server_socket = socket.socket(socket.AF_INET, socket.SOCK_STREAM)
server_socket.setsockopt(socket.SOL_SOCKET, socket.SO_REUSEADDR, 1)
server_socket.bind(('', port))
server_socket.listen(5)

client, address = server_socket.accept()
print("{} connected".format( address ))

while True:
    response = client.recv(512)
    print(response)
    time.sleep(1)

print ("close")
client.close()
stock.close()

308627993, 2022-05-10:
can you help to share your solutions ?