introlab/odas #246: Minimum number of microphones?

Jelt0, 2021-09-30:
Hi,

I was wandering if there is a minimum number of microphones necessary in order to use the ODAS system. I am looking into building a measurement mic array an channels are rather expensive… 

Thanks in advance!

SCWhite, 2021-10-05:
I think you might need at least 3 mic to become a "array" (but very inaccurate)
If you want to have Azimuth and Elevation angle, you must have at least 3 mic to form a 3D Coordinate System (xy,yz,xz)
but I guess if you just need 1 dimension data, 2 will work.

Jelt0, 2021-10-05:
Would a 3-mic-array work with ODAS? I read in one of the ManyEars-papers, which is the framework ODAS is partly based on (if I am correct) that the ManyEars system requires at least 4 Mics. 

Also, are there any restrictions on the array geometry to allow processing with ODAS? Do the mics have to be arranged in a circle, like with most of the MEMS-Arrays? Or would it be wise, to put one mic into the centre and place the three other mics on a circle around that, for a 4 Mic array for example?

StuartIanNaylor, 2022-02-22:
Hi does anyone know if a basic azimuth beamforming can be provided by a 2 mic and does anyone have a conf guide for best settings for the respeaker 2 mic?