introlab/odas #256: Missing files for make

LePtitMetalleux, 2022-03-15:
Hi, I was trying to install ODAS on my RaspberryPi but when I had to do the make command, I got an error saying that pulse/simple.h is not a file.
So I checked in the repo and couldn't find any file with this name.
It seems that you've added thoses lines : #include <pulse/simple.h>
    #include <pulse/error.h>
In your commit of February 1st
In the file [include/odas/source/src_hops.h](https://github.com/introlab/odas/commit/7183c89d00e3b69b9fc13ca7b78ce874756c5428#diff-937127c07013d7f47cd9911f5561f2e11c7f96fe191be66860980764e67d208a)
but pulse isn't existing.

I hope you can fix this quick
Thanks

FrancoisGrondin, 2022-03-15:
@philippewarren : can you look at this issue: looks like it has something to do with the latest update of the code done recently. thanks!

philippewarren, 2022-03-15:
Hello, these files are part of the pulseaudio sound server.
If the pulseaudio sound server is available on raspberry pi, you should be able to get it to compile by installing the `libpulse-dev` package or its equivalent.
I will look into clarifying this requirements and adding a way to build without the pulseaudio dependency!

LePtitMetalleux, 2022-03-16:
Indeed that was the problem. I've installed the libpulse-dev package and the installation worked.

PR #257 (2022-03-16): PulseAudio is now a CMake module, gives better diagnostics when missing
Closes #256
closed 2022-03-16 by doumdi