introlab/odas #262: Sound Source Localization with 3 Respeaker 4 Mic array

LePtitMetalleux, 2022-04-04:
Hi, this is not an issue, I just need help.

I'm working on a project of sourd source localization.
I got 3 Respeaker 4 Mic Array and using ODAS I have to find the coordinates of the source sound.

I make a TCP server to receive the data from the SSL of ODAS, so I got XYZ coordinates (Between -1 and 1 ??) and E (elevation) which is, if I understood correctly the energy.

With this I manage to find how to calculate the azimut we can get with odas_web
In [this file](https://github.com/introlab/odas_web/blob/master/resources/js/graph.js) I found out how to get an azimut in degrees.

`var az = Math.atan2(y,x) * 180 / Math.PI`

I was wondering what was the following formula :

`var inc = Math.acos(z/Math.sqrt(x*x+y*y+z*z))` which is then converted in degrees and substracted to 90

So my main problem here is to find how can I use all the data (xyz coords, energy, azimut and inc) from my 3 Respeaker 4 mic array to calculate the coordinates of the sound source.

To resume here are my questions :

- What is "inc" ?
- How can I calculate the coordinates of the sound source using the data of my 3 Respeaker 4 mic array

Thank you in advance for your help, it will be very valuable to me and sorry for all english mistakes.

- LePtitMetalleux

DmitryP11, 2022-04-13:
hi, I'm doing same thing as you

as I understand you need to have exact positions of your mics (same as position of mic in config file)
also maybe you will need to synchronize your mics (not yet sure), 
and then you will need to matchmake detected sources of sound unless you work in a totally silent room

inc should be "inclination" angle between xy plane and angle to z

btw have you considered best geometry of microphone orientation?

ELELUABDULSALAM, 2022-04-23:
var az = Math.atan2(y,x) * 180 / Math.PI is use to calculate the azimut in degree 

var inc = Math.acos(z/Math.sqrt(x*x+y*y+z*z)) is use to calculate the elevation 

I hope this help your work

rnehrboss, 2024-07-03:
I know this is old, but i wonder if anyone was successful in separation using the respeaker 4 mic.   Any pointers would be helpful.
Thanks!