introlab/odas #271: Respeaker USB not working with direct capture

atyshka, 2022-11-16:
I am attempting to use a Respeaker USB 4-mic array with ODAS, but I can't seem to get the config to work. The error I get is 
`Source hops: Cannot set channel count: Invalid argument`
I checked the config file for the USB 4 mic array, and I was surprised to see it contained 6 channels, rather than 4. However, changing it to 4 results in the same issue. Has anyone successfully used a Respeaker USB mic with ODAS?

longRookie, 2022-11-24:
https://raw.githubusercontent.com/respeaker/usb_4_mic_array/master/odas.cfg , maybe it's useful.

atyshka, 2022-11-30:
Apparently there are two versions of the respeaker firmware, one that outputs mono audio and another that outputs 6 channels. Installing the latter solved my issue.

closed 2022-11-30 by atyshka