introlab/odas #291: The channels' order of SSS do not match with SST

xiaotaw, 2023-12-14:
Hi, 

- I tested odas for a microphone array device,  and found the order of sss and sst did not match.

- I played an audio using my phone, with the phone hanging over the device, and turning around anticlockwisely.

- 4 channels of sss, and it shows the 2nd channel is the 'main' channel.

[image]

- however, in when i draw the angle = atan(y, x) of the 1st and 2nd sound source , it seems the 1st is the 'main' channel.

[image]

SeanTasker, 2024-09-03:
ODAS streams N channels (based on the config) from `sss` via `separated` or `postfiltered`. These channels can change based on activity. You need to consume the data from `sst.tracked` and take the channel to read from `sss` at any given timestamp for the configured hop size.

While a tracked source is active the channel it comes in on is usually consistent. At least it has been in my experience with a stationary array.

xiaotaw, 2024-09-19:
Thank you SeanTasker! As I drew above, `sst.tracked` outputs four channels, and each channel gives an array of info of the tracked sound source, including `id`, orientation(`x,y,z`) , and `activity`. 
```
id
x
y
z
activity
```
`sss` also outputs four channels. I assume each channel from `sst.tracked`  corresponds with those from `sss` (please correct me if I am wrong).  i.e.

```
sss_channel_1     <------>    sst.tracked_channel_1
sss_channel_2     <------>    sst.tracked_channel_2
sss_channel_3     <------>    sst.tracked_channel_3
sss_channel_4     <------>    sst.tracked_channel_4
```

So, could you explain more on how to get the right correspondence between `sst.tracked` and `sss`. 

> You need to consume the data from `sst.tracked` and take the channel to read from `sss` at any given timestamp for the configured hop size.

SeanTasker, 2024-09-19:
You are correct. They do correspond to one another but  keep in mind that the source ID can change, so they match up for each discrete hop. I.e. the timestamp in sst needs to be used to look up the hop size number of samples from sss at timestamp * hope size * channels bytes.

If a source is continuously active, I've found that it remains on the assigned channel until there is a short interruption. When starting again it can be remapped to another channel. In the screenshot above, it is possible that two sources were detected briefly and the "main" one there persisted and stayed on channel 1 (with the brief one on channel 0).

I do note that the activity feed shows index 0 where as the audio data appears on chanel 1. How are you parsing the JSON? Could it be seeing no source on channel 0 in stt and not returning information for it?

E.g. IDs
0
20
0
0

Does the JSON processor discard the 0 sources and just return you a list of what is there? The index will likely be wrong if that's the case.

xiaotaw, 2024-09-23:
Thank you SeanTasker! 

> You are correct. They do correspond to one another but keep in mind that the source ID can change, so they match up for each discrete hop. I.e. the timestamp in sst needs to be used to look up the [...]

Now I understand a continuous source will remains on the same channel with a high probability.

> I do note that the activity feed shows index 0 where as the audio data appears on chanel 1. How are you parsing the JSON? Could it be seeing no source on channel 0 in stt and not returning [...]

Here is how I deal with sst data in a json format. It's just a c++ version of 
https://github.com/introlab/odas_ros/blob/ros2/odas_ros/odas_ros/lib_odas_server_node.py

```c++
// JsonSocketServer.cc
void JsonSocketServer::SplitAndHandleJson() {
  int json_start_index = 0;
  int json_stop_index = 0;
  int depth = 0;

  for (std::size_t i = 0; i < buffer_.Size(); ++i) {
    const char& c = buffer_[i];
    if (c == '}' && depth == 0) {
      continue;
    } else if (c == '{') {
      depth++;
      if (depth == 1) {
        json_start_index = i;
      }
    } else if (c == '}') {
      depth--;
      if (depth == 0) {
        // now we have a complete struct
        json_stop_index = i + 1;

        // 将数据存入str中
        buffer_.Pop(json_start_index);
        auto json_str_length = buffer_.Pop(json_stop_index - json_start_index, const_cast<char*>(json_str_.c_str()));

        // json解析str

        std::string tmp = json_str_.substr(0, json_str_length);
        auto j = nlohmann::json::parse(tmp);

        // handle json obj
        HandleJson(j);
      }
    }
  }
}

```
```c++
// SstSocketServer.cc
// The class `SstSocketServer` is derived from the class `JsonSocketServer`
void SstSocketServer::HandleJson(const nlohmann::json& sst) {
  odas_ros_wrapper::AudioSstArray msg{};

  msg.header.seq = sst["timeStamp"];
  msg.header.stamp = ros::Time::now();
  msg.header.frame_id = frame_id_;

  for (const auto& source : sst["src"]) {
    if (source["id"] != 0) {
      odas_ros_wrapper::AudioSst sst_msg{};
      sst_msg.id = source["id"];
      sst_msg.x = source["x"];
      sst_msg.y = source["y"];
      sst_msg.z = source["z"];
      sst_msg.activity = source["activity"];
      msg.sst_array.emplace_back(sst_msg);
    }
  }

  pub_.publish(msg);
}

```

SeanTasker, 2024-09-23:
The indexes of tracked sources are aligned with the audio streams so in your example you're discarding the results to process if the source is 0. This means downstream you won't know which channel each of the results belongs to.

```
  for (const auto& source : sst["src"]) {
    if (source["id"] != 0) { <- Remove this check and handle source.id==0 downstream, or include the channel number in the message you send.
      odas_ros_wrapper::AudioSst sst_msg{};
      sst_msg.id = source["id"];
      sst_msg.x = source["x"];
      sst_msg.y = source["y"];
      sst_msg.z = source["z"];
      sst_msg.activity = source["activity"];
      msg.sst_array.emplace_back(sst_msg);
    }
  }
```
Hopefully that solves your problem!

A side note: To simplify your JSON prep code you can just look for `}\n{` as the marker to indicate that a full JSON object is ready. I've checked the source and it is consistent in outputing objects like this. This is also what the web interface does. i.e. 
```
}
{
```
I use boost to look for completed objects with `boost::algorithm::find_first(receivedData,std::string("}\n{"))` and loop until there isn't any more.

xiaotaw, 2024-09-23:
Thank you SeanTasker! It helped me a lot!

closed 2024-09-23 by xiaotaw