introlab/odas #299: Tracking static target source among different noise sources at a distance

swleungbrian, 2024-09-16:
Hi all,

I am trying odas to track static target source among different noise sources as shown below. 

[image]

Localized noises can be another person speaking next to the target source. Background noises can coherent or diffused as well.

I am using Respeaker v2 and Respeaker HAT for raspberry pi as the mic arrays and got poor separation results. 
I tried the same with distance reduced to around 30-50cm and audio separation is much better. 
I am not sure if i can tune parameters inside cfg file to fit distance extended to 5+/-2m? if yes, any suggestions on this?

Regards,

Brian

shenxiaozheng, 2024-09-20:
其实这是一个经典的问题，如果speaker距离麦克风阵列较远，那么麦克风阵列接受到的信号中，直达声的成分会较少，而主要成分是反射声和混响。反射声和混响对DOA估计是起反作用的。我认为这个库是无法解决这个问题的。在远场中使用这个库会直接歇菜。

swleungbrian, 2024-09-20:
> 其实这是一个经典的问题，如果speaker距离麦克风阵列较远，那么麦克风阵列接受到的信号中，直达声的成分会较少，而主要成分是反射声和混响。反射声和混响对DOA估计是起反作用的。我认为这个库是无法解决这个问题的。在远场中使用这个库会直接歇菜。

我有試過把 sst 設為 static, 配以已知的DOA, 但是效果很奇怪. 似乎比不自行輸入DOA效果還差很多.

shenxiaozheng, 2024-09-20:
我也试过，准确率太低。

swleungbrian, 2024-09-20:
> 我也试过，准确率太低。

那就我的應用場景, 值得投放時間去study odas 嗎? 還是自行重寫? 但我覺得如期我由零開始去試 LCMV/MVDR/GSC 那些方法. 那不如看看odas 能否改改去切合自己的應用場境. 不過odas 的代碼也要花時間去理解. 文件夾都很工整, 但就少有comments. 增加了不少理解上的難度.

shenxiaozheng, 2024-09-20:
这些方法，我全部试过，在大混响场景准确率都不高吧。