introlab/odas #306: Malformed JSON output

Rytisok, 2025-05-22:
I'm running into issues with odaslive output. I am using ReSpeaker Mic Array v2.0 with Ubuntu 24.04 LTS.
I have tried socket, file and terminal outputs, all result in just `}}}}` being returned.

```
(~/Downloads/odas/build/bin$ ./odaslive -c '/home/r/Downloads/odas/config/odaslive/respeaker_usb_4_mic_array.cfg' -v
+--------------------------------------------+
|    ODAS (Open embeddeD Audition System)    |
+--------------------------------------------+
| Author:  Francois Grondin                  |
| Email:   francois.grondin2@usherbrooke.ca  |
| Website: introlab.3it.usherbrooke.ca       |
| Version: 1.0                               |
+--------------------------------------------+
| + Initializing configurations...... [Done] |
| + Initializing objects............. [Done] |
| + Launch threads................... [Done] |
| + Threads running.................. }
}
}
}
}
}
}
}
}
}
}
}
}
}
}
}
}
```

Any help would be appreciated, thanks!

hideeee1106, 2025-06-11:
Do you have any problems solving this problem?

hideeee1106, 2025-06-12:
I am using Ubuntu 22.04, but I also encountered this issue. However, I have now resolved it.The issue lies in the snk_pots_process_format_text_json function, where sprintf(obj->buffer, "%s...", obj->buffer, ...) is used repeatedly to format a string and write it back into obj->buffer itself. This results in undefined behavior, which may lead to incorrect output or data corruption. 
The modified code is as follows：

```
void snk_pots_process_format_text_json(snk_pots_obj * obj) {
        unsigned int iPot;
        int pos = 0;

        // 安全写入，不会覆盖 obj->buffer 自身
        pos += snprintf(obj->buffer + pos, 1024 - pos, "{\n");
        pos += snprintf(obj->buffer + pos, 1024 - pos, "    \"timeStamp\": %llu,\n", obj->in->timeStamp);
        pos += snprintf(obj->buffer + pos, 1024 - pos, "    \"src\": [\n");

        for (iPot = 0; iPot < obj->nPots; iPot++) {
            pos += snprintf(obj->buffer + pos, 1024 - pos,
                            "        { \"x\": %1.3f, \"y\": %1.3f, \"z\": %1.3f, \"E\": %1.3f }",
                            obj->in->pots->array[iPot * 4 + 0],
                            obj->in->pots->array[iPot * 4 + 1],
                            obj->in->pots->array[iPot * 4 + 2],
                            obj->in->pots->array[iPot * 4 + 3]);

            if (iPot != obj->nPots - 1) {
                pos += snprintf(obj->buffer + pos, 1024 - pos, ",");
            }

            pos += snprintf(obj->buffer + pos, 1024 - pos, "\n");

            // 打印当前状态
            // printf("ipot = %d, buffer so far:\n%s\n", iPot, obj->buffer);
        }

        pos += snprintf(obj->buffer + pos, 1024 - pos, "    ]\n");
        pos += snprintf(obj->buffer + pos, 1024 - pos, "}\n");

        obj->bufferSize = 1024;

    }

```

Rytisok, 2025-06-12:
> I am using Ubuntu 22.04, but I also encountered this issue. However, I have now resolved it.The issue lies in the snk_pots_process_format_text_json function, where sprintf(obj->buffer, "%s...", [...]

yep, found PR that seems to address exactly this: https://github.com/introlab/odas/pull/293
lets hope it gets merged.